ThunderFly-aerospace/TF-G1 #17: Kryt 666_1032 nepasuje na díl  666_1025

kaklik, 2017-12-04:
- Tolerance pro nasunutí dílu 666_1032 přes kryt 666_1025 je moc malá (minimálně o 1.5mm)

Elizaa62, 2017-12-04:
Já jsem tam ten nástavec na dílu 666_1025 už nějak zmenšovala, ale asi je to zmenšení stále malé..

Elizaa62, 2017-12-05:
Vytvořila jsem dva nové díly 666_1025ZK1 a 666_1025ZK2 určené jen pro rychlý tisk a následnou kontrolu držáku.

kaklik, 2017-12-05:
Zjistil jsem že po úpravě kryt pylonu stále nepasuje do držáku na horní části kapotáže.  Příčinou byla nesprávně zmenšená část pro držák.  Zmenšení o 2mm tak nebylo rezervou pro tolerance, ale prostým celkovým zmenšením.  Upravil jsem proto zdrojový kód tak, aby držák zmenšoval správně a navíc jsem zavedl proměnnou "clearance", která umožňuje nastavit velikost tolerančního prostoru mezi díly. 
K tomu bylo potřeba pracovat s úhlem odtokové hrany a délku profilu zmenšit patřičně k velikosti tohoto úhlu.

commit d7e0eaf2 (2017-12-07): Fix  #17 nepasující kryt pylonu. Kód byl upraven pro korektní zmenšení
části na trupu. Podle úhlu odtokové hrany aerodynamického profilu.
closed 2017-12-07 by Elizaa62