ThunderFly-aerospace/TF-G1 #18: Označení typu na bočních plochách kapoty

kaklik, 2017-12-04:
Na boční plochy je třeba vyznačit typ UAV v tomto případě TF-G1

Elizaa62, 2017-12-05:
Přidán text. Akorát mám problém s písmenky. Z té jedné strany jsem text otočila, aby šel zprava doleva, akorát písmenko F je opačně tak než se píše a čte.

kaklik, 2017-12-05:
Vypadá to, že je to tím, že parametr direction = "rtl" zřejmě funguje jinak, než by se dalo očekávat, neboť neotáčí ani písmena G a ani číslici 1. 
Řešením zřejmě je text klasicky orotovat v tomto případě podle osy Y. Rovnou jsem to zkusil a font při tom změnil na PT Sans. 

Ještě je potřeba vyřešit, aby text byl na boku skutečně vidět i po vyrenderování modelu.

Elizaa62, 2017-12-25:
Písmo jsem v ose Z posunula o 0,5 mm. Nejsem si ale jistá, zda je to tak správně.

kaklik, 2017-12-25:
přidal jsem tam linear_extrude(wall_thickness), aby text neměl nulovou tloušťku. 
Teď je třeba to doladit tak, aby se text vešel do maximálního tisknutelného rozměru a nebo ho naopak udělat inverzní dovnitř. 

Aktuálně se text nepropisuje na všechny díly právě díky tomu, že některé díly se ořezávají na maximální tisknutelný rozměr.

Elizaa62, 2017-12-27:
Označení vytvořeno přes differenci. V ose z odečtení 0.3 mm.

kaklik, 2017-12-27:
Paráda :) Nápis jsem jenom posunul kousek níže, protože mi to tak připadá hezčí. 
Je to ale uzavřené.

closed 2017-12-27 by kaklik