ThunderFly-aerospace/TF-G1 #19: Šrouby pro fixaci krytu

kaklik, 2017-12-04:
Horní část kapotáže 666_1025 by měla být fixována šrouby s matkami zapuštěnými do dílu 666_1027.

Elizaa62, 2017-12-10:
Přidány díry pro šrouby do dílu 666_1025.

Elizaa62, 2017-12-10:
Upravila jsem výšku těch děr, nevšimla jsem si, že je mám každou jinak vysoko. Na té podložce pracuju.

Elizaa62, 2017-12-11:
Hotovka.

kaklik, 2017-12-12:
Myslím ze na podložce 1027 jsou ty otvory pro matky nerovnoměrně daleko od kraje. Teď se to projevilo při umisťování akumulátorů.

Elizaa62, 2017-12-12:
Myslíš otvory pro matky pro připevnění horního krytu?

kaklik, 2017-12-13:
Ano myslím, na jedné straně jsou blíže k okraji, než na druhé. Je to vidět při tomto pohledu: 

[image]

kaklik, 2017-12-25:
Zjistil jsem že otvor pro matky je příliš úzký a matka se tam svojí výškou nevejde. Zrovna v tomhle směru ale nevadí i celkem velká tolerance.  Kritický je rozměr v kolmém směru, tam matka teď pravděpodobně sedí dobře, tak akorát, aby se neprotáčela.

Elizaa62, 2017-12-25:
Otvor pro matku zvětšen na výšce matky o 1 mm.

kaklik, 2017-12-25:
Ano, otvor pro matku je zvětšen správně. :) 

Na konci otvoru je ale 6tihran, který má matku zřejmě držet. Zdá se mi, že ten zůstal stejně vysoký: 
[image]

Pro zjednodušení by možná i stačilo, kdyby na konci otvoru nebyl šestihran, ale otvor může končit obdélníkem. Zajištění matky proti otáčení na dvou stěnách by mohlo stačit.

Elizaa62, 2017-12-25:
Už je tam šestihran, musela bych to celé opravovat. Otvor pro držení matky je taky zvětšen o 1 mm.

kaklik, 2017-12-25:
Vyřešeno. :)

closed 2017-12-25 by kaklik