ThunderFly-aerospace/TF-G1 #20: Převis na přední části dílu 666_1025

kaklik, 2017-12-04:
Přední část, má dělení u výřezu pro motor příliš vzadu. V důsledku toho vzniká převis, který znemožňuje tisk. 

Taktéž boční stěny ve výřezu pro motor musí být prodlouženy až na spodní úroveň dílu, aby jejich tisk začínal od podložky. 
Je možné, že díky této úpravě bude nutné upravit i díl 666_1027

Elizaa62, 2017-12-05:
Boční stěny ve výřezu pro motor jsem prodloužila na spodní úroveň dílu, Ale překrývá se to s tím lemem. Je to tedy ve stejné úrovni, ale to by snad nemělo vadit. U toho dílu 666_1027 udělám výřez až na úroveň zářezu.

Elizaa62, 2017-12-05:
Tak jsem upravila i ten převis. Zvolila jsem špatnou hodnotu posunutí.

kaklik, 2017-12-06:
Potvrzuji, že převis je odstraněný a díl teď půjde vytisknout.. :) 
Bylo by ještě dobré, aby stěny byly přidány ne v poddružném modelu ale v hlavním modelu kapkového krytu, aby steny byly vidět i v sestavě.  Podobně jako je teď vyřešen držák krytu vertikální trubky.

Elizaa62, 2017-12-10:
Stěny přesunuty do hlavního kapkovitého modelu.

closed 2017-12-12 by kaklik