ThunderFly-aerospace/TF-G1 #22: Zámky pro přesnější lepení dílů.

kaklik, 2017-12-04:
Při lepení kapoty 666_1025 jsem zjistil, že je těžké na sebe navázat jednotlivé stěny částí. Pomohl by proto nějaký tvarový přechod, který by umožnil, aby do sebe oba díly před lepením zapadly.

Elizaa62, 2017-12-11:
Stále tento problém trvá?

Elizaa62, 2017-12-12:
Přemýšlím jak to udělat.. napadlo mě, že by se ten lem dal nahoře vykrojit a udělaly by se tam takové ploché drážky, do kterých by se nasunula a nalepila rovná podložka a ta by se pak nasunula a nalepila do stejných drážek, které by byly umístěny v protější části dílu, který by se přilepoval. Rozumíš mi trochu? :(

kaklik, 2017-12-12:
V protější části dílu by zřejmě byly spíš zuby, ne?  Myslím, že něco podobného, je tady popsané jako "spoj zubový" a nebo "pero - drážka". 

[image]

Když jsem to teď našel, tak se mi docela líbí ty "hrdlové spoje".  Možná by bylo jednodušší je vymodelovat, než ty zuby. :)

Elizaa62, 2017-12-12:
Když mě pořád trápí problém, že ty díly jsou dělené na maximální tisknutelný rozměr, takže nevím, když se tam udělá nějaký zobáček v délce navíc, tak jestli to půjde vytisknout..

kaklik, 2017-12-12:
Je výhodné, aby vnější povrch byl hladký. Takže vžechny tvarové zobáčky podobně  by bylo stejně potřeba zapustit pod povch.

Elizaa62, 2017-12-12:
Já asi moc, teda spíš určitě, nerozumím tomu, jak se to myslí.. Nedokážu si to vůbec představit :(

Elizaa62, 2017-12-13:
Kdyz tam udelam napr. Ten zubovy spoj, tak se mi celkova delka te casti zmensi. Musela bych ty zuby udelat po celem obvodu, aby to melo trn spravny efekt a nedoslo ke zvlneni. Protoze kdyz bych udelala jeden zub, tak ty vedlejsi plochy nebudou pasovat do sebe.

Elizaa62, 2017-12-13:
Co kdybch uplne zrusila ty lemy a misto tech lemu by se natiskli takove podlozky pomoci kterych by se ty dve casti jakoby podlepili?

Elizaa62, 2017-12-13:
Vytvořený pokus o zámek. Více v commitu u dílu 666_1025.

kaklik, 2017-12-13:
Ano, konstrukce zámku dobrá. Teď je to akorát potřeba udělat skoro všude..

Elizaa62, 2017-12-20:
Zámky u horního dílu krytu vytvořeny. Akorát mám problém u části B. Tys tam totiž ten lem nechal vlevo. Ale já mám u dílu A lem vpravo, takže u části B jsem lem vlevo smazala a nahradila třemi zámky. Dále u části C je lem též vlevo, tak jsem vytvořila u části B zámky i vpravo. Ale mám otázku, zda to půjde vytisknout, nebo ty lemy mám předělat tak, aby to pak šlo vytisknout?

kaklik, 2017-12-21:
Obávám se, že zámky na straně, kde by měly začínat na podložce bez lemu, který drží model na podložce nepůjde tisknout. 
Model při tisku musí  vždycky začínat lemem, aby něčím držel na podložce během tisku. 
Takže by asi bylo lepší to nějak překonstruovat.  Původně jsem tam ty lemy zvolil tak, aby se model během tisku zužoval směrem vzhůru, což mi připadalo přirozenější. Myslím ale, že se může i rozšiřovat a kvalitu tisku to neovlivní.

kaklik, 2017-12-25:
Právě jsem vymyslel, jak potíž s tiskem zámků u prostředního dílu s držákem na kryt pylonu vyřešit.  Stačí tuhle část rozdělit ještě na dva další díly, které se nebudou slepovat ale jenom k sobě budou sesazeny.  Na to stačí, když každá část bude mít svůj vlastní lem, který ji zpevní. 

Rozdělení navíc umožní aby kryt šel sundat bez toho, aby se přetahovat shora přes pylon.

Elizaa62, 2017-12-25:
Díl B jsem rozdělila na díl B a díl BB. Oba díly mají svou díru na šroub pro připevnění k podložce, svůj lem a své zámky.

kaklik, 2017-12-25:
Paráda :)  Myslím, že spojování dílů je tím vyřešené.

closed 2017-12-25 by kaklik