ThunderFly-aerospace/TF-G1 #23: Zapuštění akumulátorů do dílu 666_1027

kaklik, 2017-12-04:
Do podložky by bylo potřeba vytvořit zahloubení pro umístění dvou akumulátorů [tohoto typu](http://www.pkmodelar.cz/akumulatory-lipol-akumulatory-letecke-4-clanek-lipol/foxy-g2-li-pol-5000mah-14-8v-40-80c-74-0wh-3eb4038). 
Cílem je zajistit, aby oba akumulátory mohly být chyceny k podložce vytištěnou sponou přitaženou šroubem. A v případě potřeby je bylo možné snadno vyjmout z trupu sundáním části krytu a povolením spony.

Elizaa62, 2017-12-11:
Zahloubení pro akumálátory je vytvořeno. Hloubka zahloubení je main_tube_outer_diameter/2. Poté až vytvořím účkovou spojku ve tvaru kvádru, tak přidám otvory pro šrouby, kdy ze spodní strany podložky budou zahloubené otvory pro matky.

Elizaa62, 2017-12-11:
Objevil se problém se zahloubením.. problémem je horizontální trubka..

kaklik, 2017-12-12:
Řekl bych, že je to vyřešeno změnou pozice akumulátorů.

[image]

A teď stačí je urovnat a doplnit třmen na přitažení.

Elizaa62, 2017-12-12:
Třmenem myslíš tu mou spojku 888_1001?

kaklik, 2017-12-12:
Ano, technicky se tomu řešení myslím říká třmen: https://eshop.killich.cz/eshop/trmen-sroub-tvar-u-m10-50-bily-zinek-dn-40-22400040_22400040.html

Elizaa62, 2017-12-12:
Rozumím :)

Elizaa62, 2017-12-12:
Jinak akumulátory srovnány parametrem move_of_accumulator a zapuštěny parametrem sink_of_accumulator.

kaklik, 2017-12-13:
Paráda, vyřešeno :)

closed 2017-12-13 by kaklik