ThunderFly-aerospace/TF-G1 #24: Rozdělení spodního krytu 666_1029

kaklik, 2017-12-04:
Spodní kryt je určený jako schránka pro měřicí aparaturu. Měl by proto být uzavřeným celkem, který se namontuje na již sestavené UAV. 

Navrhoval bych jej z konstrukčních důvodů rozdělit na tři části:

1. Přední díl od začátku do první podvozkové nohy včetně. 
2. Středový uzavřený díl skládající se ze dvou polovin určených k sešroubování. 
3. Zadní díl od druhé podvozkové nohy do konce kapkového krytu. 

Přední a zadní díl může být přišroubován k podložce stejně jako vrchní kapotáž (666_1025) ale zespodu. 
Prostřední díl by pak měl být zasouvací mezi přední a zadní díl s tím, že jeho pozice by pak měla být zajištěna šrouby zašroubovanými skrz přední a zadní díl.

Elizaa62, 2017-12-25:
Díl rozdělen snad dle požadavků.

kaklik, 2017-12-25:
Rozdělení provedeno správně. :)

closed 2017-12-25 by kaklik