ThunderFly-aerospace/TF-G1 #25: Umístění GPS antény

kaklik, 2017-12-04:
Pod zadní část horní kapoty je potřeba umístit GPS anténu s magnetometrem. 
Modul vypadá přibližně takto: https://hobbyking.com/media/catalog/product/cache/1/image/320x230/9df78eab33525d08d6e5fb8d27136e95/legacy/catalog/55558s2.jpg

A je potřeba, aby byl uchycený na 3 šroubech procházejících otvory skrz plošný spoj. 

Ze spodní strany plošného spoje musí být ponecháno výškově alespoň 5mm prostoru pro součástky naletované na spodní stranu plošného spoje.

Elizaa62, 2017-12-25:
Toto má být připevněno k podložce 666_1027 v její zadní části?

kaklik, 2017-12-25:
Ano má to být co nejvíc vzadu jak se to ještě vejde.

Elizaa62, 2017-12-27:
Už jsou rozměry? :)

kaklik, 2018-01-01:
Rozměry jsou tady: 

[image]

[image]

Otvory jsou umístěny v rovnostranném trojúhelníku, kde  vzdálenost a průměr otvorů lze spočítat z předešlých fotografií měřených vzdáleností vnitřní a vnější strany otvorů. 

Vnější průměr kruhového plošného spoje 56mm. 

Protože plošný spoj obsahuje z obou stran poměrně vysoké součástky, tak by zřejmě bylo potřeba, aby otvory byly opatřeny distančními sloupky vysokými alespoň 8mm.

Elizaa62, 2018-01-04:
Nakonec se mi tam podařilo dát všechny tři sloupky mimo horizontální trubku.

kaklik, 2018-01-19:
Anténa na podložku pasuje. Zatím se úkol může uzavřít dokud nevyzkouším že se vejde pod kryt.

closed 2018-01-19 by kaklik