ThunderFly-aerospace/TF-G1 #26: Umístění OpticalFlow senzoru

kaklik, 2017-12-04:
Do zadní části spodního kritu je potřeba připravit držák Optical Flow senzoru a připravit i průzory pro kameru a ultrazvukový výškoměr. 

OpticalFlow senzor vypadá takhle: https://ae01.alicdn.com/kf/HTB1toBjRXXXXXXRXpXXq6xXFXXXd/PX4FLOW-V1-3-1-Optical-Flow-Sensor-Smart-Camera-with-MB1043.jpg

Jeho úkolem je koukat dolu na zem a měřit vzdálenost od země a rychlost pohybu vůči zemi. Obě jeho čidla proto musí mít dobrý výhled na zem. 

Rozměry OpticalFlow senzoru](https://pixhawk.org/_media/modules/px4flow-manual-v1.3.pdf).

Elizaa62, 2017-12-25:
Ten držák má být tisknutelný zvlášť a poté připevněn šrouby ke spodnímu krytu?

kaklik, 2017-12-25:
Napsal jsem to nepřesně. Ten senzor má držek na podložce v její zadní části zespodu.  Není na to myslím potřeba vytvářet zvlášť tisknutelný díl. 
Ve spodním krytu pak budou pouze kulaté otvory, pro čidla.

Elizaa62, 2017-12-27:
K čemu tam jsou ty šrouby M2? Mají také procházet skrz podložku?

kaklik, 2017-12-27:
Šrouby M2 slouží pro připevnění objektivu.  Nebudou proto do podložky vůbec zasahovat. 
To čidlo stačí když bude držet šrouby, které budou procházet těmi otvory v rozích.

Elizaa62, 2017-12-27:
Nápad.. Spojka 666_1004, která spojuje trubku horizontální a trubky od podvozku, je umístěna v zadní části podložky. Nedala by se tato spojka modifikovat tak, že by se ke spodní části spojky natiskla placka, ke které by se poté přidělal OpticalFlow senzor?

kaklik, 2017-12-27:
Skvělý nápad. :)  Ještě bych byl nejraději, kdyby ta placka byla nějakým způsobem odnímatelná od spojky jako díl. Pro tento požadavek mám tyto důvody: 

1. Kdyby se z nějakého důvodu stalo, že se držák optical flow senzoru poškodí, tak by bylo potřeba vyměnit celou spojku, což teď znamená rozebrat v podstatě celou konstrukci, vyměnit spojku a pak to celé dát znovu dohromady. 
2. Tisk placky může ovlivnit tisknutelnost a kvalitu celé spojky
3. Může se stát, že spojky budou tisknuté z nějakého speciálního pevnostního plastu a nedávalo by smysl tisknout z něj nástavec pro opticalflow senzor. 
4. Není moc praktické mít různé díly pro přední a zadní spojku, obě spojky ale mohou mít nějakou přípravu pro uchycení dalších dílů.

Elizaa62, 2017-12-27:
[image]

To černé značí původní spojku 666_1004. To červené značí, jak by se modifikovala ta spojka, aby pod tou plackou byla větší styčná plocha. To zelené značí odjímatelnou placku, ke které se přidělá Optical Flow senzor. Placka bude mít dvě "nohy" které budou široké jako ta spojka a ke spojce budou přidělány stejným šroubem, který spojuje samotnou spojku a trubku podvozku. Fungovalo by to?

kaklik, 2017-12-27:
Jo, to by mohlo fungovat! :) Když se tam dají delší šrouby, tak ta placka půjde přitáhnout matkou na přečnívající šroub. A ani tam pak nemusí být ta druhá noha na placce. 
V podstatě se ta placka může proti otáčení pod šroubem zajistit o to zkosení spojky, tím že na placce bude ten materiál (červený), který na spojce chybí. :)

Elizaa62, 2017-12-28:
On by se pak teoreticky mohl dat sroub jeste do toho mista zkoseni. Mam totiz strach, ze by to ta jedna strana nemusela vydrzet. Ale to by se pak muselo resit podle toho, jak bude umistem ten senzor na te placce a kde budou umisteny srouby.

kaklik, 2017-12-28:
Ano, do místa zkosení dát třeba kapsu na matku by mohlo fungovat taky :) 
Ten senzor ale váží asi 20g, takže největší riziko poškození je způsobené tím, že si někdo při montáži nedá pozor a něčím do toho drbne. Pro tento případ by bylo právě vhodné, aby když už to praskne, tak aby praskl nějaký díl, který se dá relativně snadno vyměnit.

Elizaa62, 2017-12-29:
Vytvořen díl 888_1003. Je konstruován tak, že se nejdříve připevní OpticalFlow senzor a celé se to připevnění pomocí prodlouženého šroubu M3 ke spojce 666_1004.

kaklik, 2018-01-04:
Zkusil jsem senzor našroubovat na vytištěný držák a šrouby pasují! :)

[image]

Pak jsem taky změřil výšku jednotlivých senzorů na desce.  Výška je měřená od roviny tištěného modelu, na kterou je senzor našroubovaný. 

[image]

[image]

kaklik, 2018-01-19:
Bylo by vhodné držák přesunout mimo spojku do zadní části spodního krytu. Neboť výška mezi krytem a spojkou zřejmě není dostatečná pro umístění senzoru.

Elizaa62, 2018-04-17:
Vytvořila jsem držák na optical flow senzor. Je to díl 888_1003_print v souboru 888_1003.scad.  Držák bude umístěn v zadní části spodního krytu jako je na obrázku. Bude ke krytu přišroubovaný. Pokud tento pokus schválíš, tak pak ještě vytvořím otvory pro šrouby a otvory pro samotný senzor. 

[image]

Elizaa62, 2018-06-26:
Jak jsme na tom s tímto dílem?

kaklik, 2018-06-26:
Myslel jsem si, že ten držák plánuješ upravit tak, aby držel ze spodku na podložce. Alespoň tak jsem chápal tvůj poslední komentář, kde jsi si tu úlohu zároveň přiřadila na sebe. 
A já s takovým řešením souhlasím. Myslím že bude lepší, když ten senzor bude držet na podložce a ne na krytu, protože kdyby držel na krytu, tak při sundání krytu bude kryt stále držet za dráty od senzoru.   Což by asi mohlo kde koho při rozebírání překvapit.

closed 2019-10-15 by kaklik