ThunderFly-aerospace/TF-G1 #27: Umístění Pitotovo trubice

kaklik, 2017-12-04:
Úpravou dílu 666_1014 je potřeba vytvořit držák "airspeed" senzoru na zadní podvozkovou nohu. 
Snímač vypadá takto: https://ae01.alicdn.com/kf/HTB1IUPMKpXXXXXMXpXXq6xXFXXXg/Pixhawk-PX4-Differential-Airspeed-Head-Pitot-Tube-PITOT-Tube-Airspeedometer-for-PX4-Autopilot-Flight-Controller.jpg_640x640.jpg

A jeho úkolem je měřit rychlost proudění vzduchu.  K tomu je potřeba, aby měřící trubice směřovala dopředu proti ose letu.

kaklik, 2018-04-13:
Aby se tenhle úkol také pohnul, tak jsem vymyslel ještě alternativní, možná jednodušší řešení, že by se pitotova trubice dala do vrchní části krytu 666_1032. Do úrovně rotorové hlavy. 
V krytu by bylo možná jednodušší vytvořit potřebný otvor a místo se šroubem kterým se pitotova trubice přišroubuje. 

Výhodou tohoto řešení je, že by vstupní otvory byly lépe chráněny před znečištěním.

Elizaa62, 2018-04-16:
Já si asi nedokážu pořádně představit, jak to myslíš...  protože když na dílu 666_1032 udělám otvor pro šroub pro připevnění trubice, tak mi ten díl pod tíhou šroubového spojení praskne nebo se ohne..

kaklik, 2018-04-16:
Tady jsem zkusil změřit některé významné rozměry té sestavy trubice. 

Vnější průměr trubky
[image]

Vzdálenost okraje trubky k hraně díry pro připevnění.
[image]

Elizaa62, 2018-04-16:
Vytvořeno připevnění k dílu 666_1032.

kaklik, 2018-04-17:
Ano, tak nějak by to mohlo být. 

- Bylo by ale potřeba tu trubici vysunout co nejvíc dopředu.  Protože na ty dva vývody se připojují hadičky. A takhle tam bude málo místa na ohyb. 
- Šroub pro uchycení je umístěný v jiné úrovni, než samotná trubka. Viz poslední fotografie v mém komentáři.

kaklik, 2018-04-27:
Tak myslím, že ten nápad funguje docela dobře. :) 

Potenciální problém vidím v ostrém ohybu hadičky. Ale zatím to tak necháme. 

[image]

closed 2018-04-27 by kaklik