ThunderFly-aerospace/TF-G1 #30: Dělení horního krytu 666_1025

kaklik, 2017-12-06:
Při tisku tohoto dílu: https://github.com/kaklik/Components/blob/master/STL/print_plate3.stl jsem zjistil, že držák krytu vertikální trubky je ve skutěčnosti horizontální převis a kazí se tím kvalita tisku. 

Řešením by bylo posunout dělení částí krytu tak, aby držák začínal na podložce a tisková podložka ho tak od začátku tisku podpírala.

Elizaa62, 2017-12-12:
Mě ten odkaz nejde otevřít..

Elizaa62, 2017-12-20:
Není toto už náhodou vyřešené?

kaklik, 2017-12-21:
Omlouvám se za nefungující odkaz, soubor jsem smazal a neuvědomil si to. Odkaz nyní funguje.  Vyřešené to ale není, neboť díl vypadá takhle: 

[image]

Tím pádem se začátek držáku krytu pylonu stále tiskne do vzduchu.

Elizaa62, 2017-12-25:
Vyřešeno rozdělením tohoto kusu na díl B a BB a přidání lemu ke každému kusu.

kaklik, 2017-12-25:
Ano, je to vyřešeno skoro úplně.  Teď akorát vznikla nová potíž s tím, že vnikl díl, který se při tisku rozpadne na dvě samostatné poloviny.  Na screenshotu uprostřed: 
[image]

Řešením je dělení krytu poposouvat tak, aby po vytištění byly obě části spojené a zároveň aby se délka mezi děleními vešla do maximálního tiskového rozměru. Tiskárny edee: https://www.ysoft.com/cs/edee/edee-printer/edee-printer-detail

Elizaa62, 2017-12-27:
[image]

Elizaa62, 2017-12-27:
Takhle by to dělení už mohlo být?
Ptám se protože se mi nechce stokrát vše předělávat.

kaklik, 2017-12-27:
ano, takhle to být může. :) 
Tomu stokrát předělávání se dá vyhnout i tak, že místo toho dělení bude nastavitelné. 
Ale rozdělení z obrázku mi momentálně bude stačit.

Elizaa62, 2017-12-27:
Přestávám tě mít ráda a začínám se propadat do další deprese...

Elizaa62, 2017-12-28:
Zavedení nové proměnné pro dělení, proměnná zakomponována v rozdělení pro tisk a v umístění šroubů, rantlů a zámků.

kaklik, 2017-12-28:
Ta změna na parametrické dělení je úplně parádní. Rovnou jsem to i využil a snížil jsem o jeden počet dílů krytu tím, že jsem posunul dělení tak, aby lépe vycházela do tiskového rozměru. 

[image]

closed 2017-12-28 by kaklik