ThunderFly-aerospace/TF-G1 #33: Tloušťka podložky

kaklik, 2017-12-07:
Tloušťka podložky 666_1027 by se měla odvíjet od parametru main_tube_diameter.

Elizaa62, 2017-12-11:
Intersection pomocí cube ve velikosti v ose y -> main_tube_diameter

kaklik, 2017-12-25:
Z důvodu snížení hmotnosti by zřejmě šlo na mnoha místech podložky odečíst kvádry, aby nebyl žádný materiál na místech, kde to není potřeba.  To by zřejmě bylo vhodné udělat jak ve svislém, tak i v podélném i v příčném směru.  Hmotnost současného modelu podložky je cca 260g.

kaklik, 2017-12-28:
Vytvořil jsem první pokus o odlehčení podložky. 

[image]

kaklik, 2018-01-12:
Tady teď asi ještě vznikla potíž tím, že jsem odebral zakulacení hran podložky. Tím se změnila celkové tloušťka. 
Zřejmě by to díky tomu chtělo znovu nastavit tloušťku podložky tak, aby spojky na hlaví trubce byly zároveň s rovinou podložky. A aby spodní plocha podložky právě dosedala shora na trubky od podvozkových nohou.

kaklik, 2018-01-16:
Tak tloušťku spojky jsem změnil tak, aby kopírovala parametry spojky 1004.

commit 0fdcc488 (2018-01-16): Uprava rozměrů tloušťky podložky. Nyní by měla pasovat na výšku spojky pro podvozkové nohy.  FIX #33
closed 2018-01-16 by Elizaa62