ThunderFly-aerospace/TF-G1 #36: Připevnění horizontální trubky k dílu 666_1027

Elizaa62, 2017-12-12:
Původně jsem myslela, že by se ta trubka dala k podložce připevnit účkovou spojkou, které by se ale musela připevňovat šrouby, což je váha navíc. Ale jelikož trubka je v té podložce položená hluboko, tak mě napadlo, že když plast, ze kterého se tisknou veškeré věci, je ohebný, tak by se mohly vytisknout takové podložky (viz díl 888_1002.scad), které by fungovali na principu připevňování horního a spodního dílu krytu. Jakoby by se zacvakly do připravených děr v otvoru na horizontální trubku. V dílu 666_1027 jsem otvory na tyto spojky znázornila. Jsou pod bodem //připevnění k horizontální trubce. Mohl by ses prosím vyjádřit k tomu, zda by to fungovalo?

kaklik, 2017-12-13:
Nerozumím tomu, jak to má fungovat. V dílu 666_1027 otvory nevidím. Vidím jenom tohle: 

[image]

kaklik, 2017-12-22:
Asi bych navrhoval řešení, že se díl připojí ke trubce podobně jako jiné vytištěné spojky.  A to tak, že se zboku do trubky provrtá otvor třeba na 50mm dlouhé M3 šrouby. Které budou zapuštěné v podložce.  Počet šroubů nemusí být velký mohl by stačit jeden do každého  tištěného segmentu podložky.

Elizaa62, 2017-12-25:
Vytvořila jsem tam otvory zároveň pro šroub a matku v každé dělené části v hloubce main_tube_outer_diameter/2.

kaklik, 2017-12-25:
Umístění šroubů je úplně tak jak jsem to myslel.  Jenom vidím, že šrouby jsou ze spodní strany odkryté:
[image]

Myslím, že by bylo lepší, kdyby šrouby byly zapuštěné podobným způsobem jako šrouby pro připevnění třmenu akumulátoru.

Elizaa62, 2017-12-25:
Už neprotínají plastové spojky, posunula jsem je. A jak tam chceš teoreticky vůbec ten šroub vložit, když ze spodní strany bude zakrytý?

kaklik, 2017-12-27:
Vložení šroubu bude z boku šetihrannou dírou. Z druhé strany se vloží matka. Udělal jsem vzorovou úpravu na předním dílu.

Elizaa62, 2018-01-04:
Opraveno dle tvého způsobu.

kaklik, 2018-01-04:
Jo tak jsem si to představoval.  Jenom teď bude potřeba vymyslet, jak to ještě zkombinovat s odlehčením podložky.  
Zřejmě v těch místech šroubů bude potřeba po odlehčení přidat materiál, tam kde je potřeba. 

Jinak tohle issue nedovedu zavřít, protože není moje a nemám tlačítko "close"

closed 2018-01-04 by Elizaa62