ThunderFly-aerospace/TF-G1 #37: Vytvoření děr pro připevnění krytu 666_1029

Elizaa62, 2017-12-12:
Vytvořit díry pro šrouby ve spodní části krytu, abychom ho mohli připevnit k podlože 666_1027.

kaklik, 2017-12-13:
Je ale potřeba aby to bylo vyřešené trochu jinak viz #24

Elizaa62, 2017-12-25:
Koukni na ty díry, zda tak v té přední části mohou být.

kaklik, 2017-12-25:
Nejsem si jistý, jestli to bude fungovat, ale je to potřeba vyzkoušet. 

Ještě je potřeba vymyslet šrouby pro středovou část krytu, tam bych chtěl, aby se kryt připevňoval k příčným trubkám pro podvozkové nohy. Neboť uvnitř krytu bude zhruba 1kg nákladu. 

Ty dva díly středové části spodního krytu by měly být samonosný sešroubovaný celek, který má sloužit jako výměnný kontejner na náklad.

Elizaa62, 2017-12-26:
Myslela jsem, že ty dva prostřední díly by mohly být spojeny šrouby skrz lemy. Že bych udělala širší lem a do toho bych udělala třeba čtyři otvory pro šrouby, pomocí kterých by se ty dva díly (B a C) spojily k sobě. A z druhé strany bych udělala ty zámky. Pak jsem také přemýšlela, že bych na dílech spodního krytu udělala zámky, které by se zaklesly do horního krytu, aby bylo zajištěno správné nasazení krytu.

kaklik, 2017-12-26:
Spojit díly B a C přes něco jako lem je dobrý nápad. Ty dva díly ale budou ze všech stran uzavřené, což znamená že spojovací šrouby musí být dostupné z venku. Moje představa proto vypadala tak, že by hlavičky šroubů by byly dostupné otvorem z dílu C. Podobně jako na tomto obrázku: 
[image]
A uvnitř dílu B by byly kapsy s matkami.

Na všech vnějších stranách dílů B a C pak budou plné stěny, včetně "stropu". Protože středová část B a C musí fungovat jako samostatný uzavřený kontejner na náklad.  Zámky tam ale být mohou také.

kaklik, 2017-12-27:
Upravil jsem díly B a C, aby se více podobaly uzavřenému kontejneru, který je potřeba.

Elizaa62, 2017-12-29:
Já jsem to teď asi rozhašila..

kaklik, 2017-12-29:
Mám to nějak opravit?

Elizaa62, 2018-01-30:
Přidala jsem otvory pro sešroubování dílu 2 a 3.

kaklik, 2018-01-30:
Jo úplně něco takového jsem myslel. :)  Teď je ještě potřeba zajistit, aby díly 2 a 3 byly uzavřený celek.

Elizaa62, 2018-01-31:
Vytvořila jsem ten uzavřený kontejner. Přidala jsem tam ještě nějaké čtverce ke slepení, ale nejsem si jistá, zda to půjde vytisknout.

kaklik, 2018-01-31:
Myslím že se  takhle uzavřené to vytisknout půjde.  Akorát mi připadá, že ta horní stěna ale má jinou tloušťku než ostatní stěny. 

Co se myslí přidáním čtverců ke slepení?  Protože na těchto dílech 2 a 3 nemají třeba ty zámky co tam teď jsou moc význam. Protože se k ničemu lepit nebudou.   Díly 2 a 3 mají být vyměňovací na již sestaveném kompletu.

Elizaa62, 2018-02-05:
Ty zámky jsem myslela jen, aby to zapadlo do sebe. Ale je pravda, že u dílu 2 a 3 moc veliký význam nemají.

Elizaa62, 2018-02-05:
Snažila jsem se nechat zámky jen mezi díly 4 a 5, ale nejde mi to. Asi nerozumím tomu kódu, cos vytvořil na tolik, abych to uměla změnit, tak jak potřebuji. Ani po třech hodinách zkoumání a vymýšlení mi to nejde.. Promiň...

kaklik, 2018-02-05:
Asi není úplně nutné, aby jsi tím trávila hodiny. Spíš by bylo užitečnější mi buď dát vědět, že se o to snažíš a nejde ti to. A nebo se rovnou zeptat, jak by to šlo udělat.  Smazat zámky jenom mezi některými díly v tom současném řešení není úplně jednoduché a vím o tom, je tam třeba zavést podmínku, mezi kterými díly mají zámky být. 
Nevím ale jestli jsi to třeba nezkoušela udělat.

Elizaa62, 2018-02-05:
Podmínku jsem nezkoušela zavést, ani to neumím..

kaklik, 2018-02-06:
V tom modelu spodního krytu, je ten univerzální způsob dělení na části (který je použitý v horním krytu) spíš komplikace.  
Protože spodní kryt bude mít téměř každé dělení speciální.   Například díly 2 a 3  (B a C v původním označení) budou mít úplně spociální dělení se šrouby.  Díly, které na ně navazují 1 a 4 budou taky dělené speciálně, protože  blok z dílů 2 a 3 bude muset jít celý vyndat ze složeného modelu.  Těch speciálností je tam hodně... 
Asi bych navrhoval nechat tam to rozdělení na díly A, B, C, D, E. Protože to tak v tomhle případě bude nejpřehlednější.

Elizaa62, 2018-02-10:
Nechala jsem rozdělení na A,B,C,D,E. Zámky jsem dala jen mezi díly D a E, přičemž zámky jsou  na dílu D a na dílu E je rantl a otvory pro zámky. Díl A je úplně bez rantlu.

Elizaa62, 2018-02-11:
Můžu toto issue uzavřít?

kaklik, 2018-02-11:
Ano, tohle je myslím vyřešené.

closed 2018-02-11 by Elizaa62