ThunderFly-aerospace/TF-G1 #42: Zrychlení vykreslování

kaklik, 2017-12-18:
Myslím, že tady by klidně šlo ty zámky zarovat s kapkou všechny najednou a ne po jednom. Takhle se musí vždycky počítat s celou kapkou i když se zarovnává jenom malý kousek. 

https://github.com/Elizaa62/Components/blob/5ee8c31934eb539f08c21216f764b11d9fb347ce/src/666_1025.scad#L300-L341

Elizaa62, 2017-12-19:
Asi ti nerozumim.. :(

kaklik, 2017-12-19:
Tenhle kód: 
```
union(){ 
   intersection(){ 
      whole_drop(); 
     //čtverec pro zámek 
      translate([49,2,hull_z_size/2-10]) 
         rotate([0,0,0]) 
              cube([hull_wall_thickness*2,15,10]); 
}
```

Udělá to, že umístí kostku  a tu pak zvenku ořízne podle kapky.   Protože je to ale výpočetně poměrně náročná operace  která se provádí několikrát po sobě.  Tak by to šlo vylepšit tak že se umístí všechny kostky, vytvoří se z nich union a to celé se pak najednou ořízne pomocí intersection.

Elizaa62, 2017-12-20:
Tak jsem to tak  udělala a opravdu se to zrychlilo :)

closed 2017-12-21 by kaklik