ThunderFly-aerospace/TF-G1 #44: Velikost prostoru pro akumulátory

kaklik, 2017-12-25:
Výška třmenu pro připevnění akumulátorů zřejmě nepočítá s částečným zapuštěním akumulátoru do podložky. 

[image]

Zároveň skutečné rozměry akumulátoru jsou zřejmě jiné než deklarované v dokumentaci.

[image]
[image]
[image]

Elizaa62, 2017-12-25:
Rozměry otvorů pro akumulátory jsem dala přesně takové jaké jsou zobrazeny na posuvce.

Elizaa62, 2017-12-25:
Rozměr spojky nevím proč nepočítal nepočítal se zahloubením, ale už by to mohlo snad fungovat.

kaklik, 2017-12-25:
Díval jsem se, že rozměry akumulátorů jsou nastaveny jednak v module 888_1001() a pak také v module 666_1027(). Aby se správně počítalo zahloubení, tak je potřeba, aby parametry byly nastavené jenom na jednom společném místě a to nejspíše v souboru Parameters.scad. 
Chtěl jsem to přesunout, ale nevím co je proměnná "okraj" takže jsem to nechtěl rozvrtat.

Elizaa62, 2017-12-27:
Všechny parametry týkající se akumulátoru jsem přesunula do parameters.scad. Jsou tam nominální rozměry akumulátoru, jeho posunutí na podložce a výška zahloubení.

kaklik, 2017-12-27:
Takže vyřešené.

closed 2017-12-27 by kaklik