ThunderFly-aerospace/TF-G1 #45: Zub na lemu krytu

kaklik, 2017-12-25:
Na horním krytu vznikl během úprav zub, který je mezi spodním lemem patřícím do drážky v podložce  původním krytem bez lemu: 
[image]

Elizaa62, 2017-12-27:
Nerozumím...

kaklik, 2017-12-27:
Na přiloženém obrázku je vidět, že spodní lem nenavazuje hladce na stěnu, ale je tam zub.  Domnívám se, že je to způsobené tím, že spodní lem je nějak doplněný na předtím již zakulacené zakončení stěny.

Elizaa62, 2017-12-27:
Ale jak to mám udělat?

Elizaa62, 2017-12-27:
Vypadá, že ten zub nezpůsobuje ten lem.

Elizaa62, 2017-12-27:
Tak jsem tam nechala ten můj způsob. Jen jsem změnila odečítací kapku.

kaklik, 2017-12-27:
Nevím co přesně se změnilo, ale nyní je zub u stěny v přední části krytu: 
[image]

Na lemu zub ale už nevidím. :)

Elizaa62, 2017-12-27:
Tak změnil se ten zub z těch předchozích obrázků.

kaklik, 2017-12-27:
Ano, ten tam už není. Myslel jsem to tak, že nevím co způsobilo, že je tam tenhle nový zub, který tam myslím nebyl.

Elizaa62, 2017-12-27:
Už by tam ten zub být neměl.

closed 2018-01-04 by kaklik