ThunderFly-aerospace/TF-G1 #50: Vytvořit otvory pro trubky od podvozku v 666_1027

Elizaa62, 2017-12-27:


kaklik, 2018-01-09:
Tady nerozumím co se tím myslí.

Elizaa62, 2018-01-09:
myslí se tím to, že si myslím, že jelikož se dělají díry do spodního krytu na trubky a ten spodní kryt zasahuje až do půlky té podložky, tak v té podložce budou taky potřeba otvory na ty podvozkové trubky.

kaklik, 2018-01-09:
Takhle jak to je nakreslené, tak když se podložka nasadí na středovou trubku, tak přesně dosedá jak na středovou trubku, tak i na shora příčné trubky k podvozkovým nohám. Myslel jsem, že je to tak záměr.

Elizaa62, 2018-01-09:
Asi je. Nekoukala jsem, jak to tam vychází.

kaklik, 2018-01-09:
V tom případě si myslím, že to vychází dobře a prozatím se to může uzavřít, aby to neodvádělo pozornost od důležitějších úkolů.

commit 68c863e3 (2018-01-16): Příprava pro tisk.

CLOSE #50
closed 2018-01-16 by Elizaa62