ThunderFly-aerospace/TF-G1 #56: 666_1029

Elizaa62, 2017-12-29:
Vytvořila jsem pro dělení spodního krytu globální parametry jako jsou pro hodní kryt. Zde je to proměnná "bottom_cover_division". Požadoval jsi dělení za přední podvozkovou nohou a před zadní podvozkovou nohou. Jelikož umístění podvozkových noh mám určené zdlouhavým vzorcem, tak jsem si hodnotu spočítala podle jmenovitých hodnot a dělení provedla přibližné, aby to vycházelo. Mám akorát strach v části D, kde je před zadní podvozkovou nohou zbývá malé množství materiálu, aby nedocházelo k ulamování. Důvodem tohoto malého množství materiálu je to, že kdybych dělení posunula, tak by musela vzniknout nová část, protože maximální tisknutelný rozměr by nepojmul celou zbylou část dělenou před zadní podvozkovou nohou až do konce krytu.

Dále podle této globální proměnné pro dělení jsem vytvořila umístění šroubů a rantlů. 

Zámky zatím hotové nejsou.

V části B a C jsem tloušťku rantlu udělala 4*hull_wall_thickness. Důvodem je vytvoření otvoru pro šroub a matku pro spojení těchto dvou částí. Zatím jsou zde jen dva šrouby. Mám případně ještě doplnit dva šrouby do spodní části rantlů?

kaklik, 2017-12-30:
Podíval jsem se na to, myslím že řešením je to, že poslední koncový díl teď ve skutečnosti není dlouhý přes celý tiskový rozměr. Protože jeho délka se počítá z délky celé kapky, ale reálně je to pouze menší část vyříznutá ze spodní strany, nikoli z celé délky. 
Takže to dělení se může určitě posunout. Minimálně tak, aby to bylo symetrické s tím předním dílem.

kaklik, 2017-12-30:
Všiml jsem si, že část B a C jsou spojeny tak, že sešroubován rantl na vnitřní straně.  Tento způsob spojení ale nepůjde použít, protože ke šroubu nebude možné se dostat. 
Části B a C mají tvořit uzavřený kontejner, kde budou zakryty všechny strany. Je proto potřeba použít nějaký typ spojení, který bude přístupný zvenku.  Jeden možný způsob jsem již navrhoval v #37 .

Elizaa62, 2018-01-02:
Maximální tisknutelný rozměr je 150 x 150 mm. Tím, že jsem si tam zobrazila kvádr o těchto maximálních tisknutelných rozměrech, se mi zobrazilo, že ten konec musím rozdělit na dvě menší části, abych před tou zadní podvozkovou trubkou měla více materiálu. Ale o tom jsem psala už nahoře myslím.

Elizaa62, 2018-01-02:
Jak myslíš to, že na dílu B a C budou plné stěny včetně stropu?

kaklik, 2018-01-02:
Aha, dobře už to vidím taky, ta rezerva je tam zanedbatelná.

[image]

Takže bude asi nutné to rozdělit na dva díly.  Protože jinak se tam zřejmě nevejde ani ten držák OpticalFlow senzoru. 

Plnými stěnami myslím to, že ten díl bude dutý ale zvenku všude zakrytý. Prostě vodotěsná nádoba.

Elizaa62, 2018-01-05:
Poslední část rozdělena na dvě. Dále vytvořeny otvory pro šrouby pro připevnění k podložce. Otvory jsou v části A, D a E. Otvory pro spojení částí B a C zatím nejsou vytvořené.

closed 2018-01-31 by Elizaa62