ThunderFly-aerospace/TF-G1 #58: Spasování lemu na krytu a drážky na podložce

kaklik, 2018-01-09:
Lem na krytu nezapadne do drážky v podložce, protože tomu brání vnitřní lemi na spojích krytu. 

[image]

[image]

Elizaa62, 2018-01-25:
Co jsem pochopila, tak podložka je nahoru zvětšena o coupling_wall_thickness, tudiž jsem rantly posunula též o tuto hodnotu a plus jsem přidala 2*global_clearance.

kaklik, 2018-01-26:
Je tam v nějakém modelu provedený test toho, že to do sebe pasuje?

Elizaa62, 2018-01-26:
Zkoumala jsem to akorát tak, že v souboru 666_1027 jsem si zobrazila horní kryt a detailně jsem na to koukla.

kaklik, 2018-01-26:
Dobrá, zřejmě je  ještě potřeba posunout pozici zámků: 

[image]

Elizaa62, 2018-01-27:
Mělo by být opraveno.

closed 2018-01-30 by kaklik