ThunderFly-aerospace/TF-G1 #62: Odstranění magických konstant z modelů

kaklik, 2018-01-19:
Některé modely obsahují magické konstanty, které jsou ve skutečnosti závislé na jiných parametrech ale nyní mají hodnotu takovou, aby to zrovna vyšlo. 
Příkladem je třeba model držáku akumulátoru.   Když jsem změnil parametry rozměrů akumulátoru, tak model přestal fungovat a není to vůbec jednoduché opravit.

Elizaa62, 2018-01-25:
Pokud to teď může být tak jak to je, tak to budu řešit až později.

kaklik, 2018-01-26:
Klidně, to není kritický problém. Momentálně to nejvíc vadí jenom u držáku akumulátoru z issue #65

Elizaa62, 2018-02-10:
Myslím si, že je to vyřešeno. Co myslíš?

kaklik, 2018-02-10:
Tohle bych asi ještě nezavíral. Navrhuji, abych sem dopisoval konkrétní divnosti co najdu.  
Například teď jsem zjistil, že pozice držáku krytu pilonu, je v horním krytu zadaná číselně na několika místech. 
Jedno místo je například tady: https://github.com/Elizaa62/Components/blob/master/src/666_1025.scad#L241

Správně by to zřejmě mělo být svázané s pozicí prostřední spojky na vertikální trubku.  Což asi má být někde v Parameters.scad , pokud to tam už není.

Elizaa62, 2018-02-14:
Tento problém jsem myslím vyřešila zavedením globální proměnné cover_pilon_position.

kaklik, 2018-02-21:
Ok, pozice pylonu je vyřešená.

kaklik, 2018-02-21:
Po sestavení celého modelu jsem si také všiml, že díry na šrouby v krytech jsou občas navíc. Zřejmě by proto bylo dobré, kdyby pozice šroubů byly převzaty z nějaké společné tabulky, podobně jako dělení dílů. 
Teď je to řešeno násobky z pozic dělení, což není úplně špatně, ale je na tom blbé to, že ty násobky musí být zapsané v každém modelu stejně, jinak to nefunguje.   Lepší by proto bylo si pozice děr poředpočítat z ostatních parametrů třeba přímo v souboru parameters.scad

kaklik, 2018-03-08:
Pozice děr v ose X jsou myslim vyřešené. Teď by chtělo ještě odstranit konstanty, které udávají vzdálenost kapes pro matky od stěny. 
Je to zřejmě poměrně složitá úprava stejně jako odlehčení, ale umožnilo by to měnit velikost podložky bez toho, aby se musely ručně posouvat díry na šrouby.

Elizaa62, 2019-02-02:
> Ok, pozice pylonu je v horním krytu vyřešená. Ještě to ale chybí udělat v podložce.

Kubo, koukám tady na tento komentář. Existují dvě proměnné... "Main_pilon_position" , kdy tento parametr určuje pozici vertikální trubky, tím i pozici prostřední spojky a je využit v podložce. Parametr "Cover_pilon_position" určuje pouze pozici krytu pylonu a je využit v horním krytu. Každý z těchto dvou parametrů má tedy svou samostatnou hodnotu a nenavazují na sebe. Pokud si vzpomínám, tak hodnotu parametru "Main_pilon_position" jsem brala z výkresu. A hodnotu "cover_pilon_position" jsem brala dle šířky spojky 666_1026, která je umístěna v horní části vertikální trubky.  Otázkou je, pokud chceš "cover_pilon_position" navázat na pozici prostřední spojky na vertikální trubku, jakou hodnotu tam dát. 

A teď nastává ještě jedna otázka.. Vzhledem k tomu, že se spojka v horní části vertikální trubky změnila, bude se muset modifikovat i kryt pylonu (666_1032), že?

Elizaa62, 2019-02-02:
> Pozice děr v ose X jsou myslim vyřešené. Teď by chtělo ještě odstranit konstanty, které udávají vzdálenost kapes pro matky od stěny.
> Je to zřejmě poměrně složitá úprava stejně jako odlehčení, ale umožnilo by to měnit velikost podložky bez toho, aby se musely ručně posouvat díry na šrouby.

Řešilo se toto nějak...?

Elizaa62, 2019-02-02:
> Po sestavení celého modelu jsem si také všiml, že díry na šrouby v krytech jsou občas navíc. Zřejmě by proto bylo dobré, kdyby pozice šroubů byly převzaty z nějaké společné tabulky, podobně jako [...]

Toto je vyřešené, že?

Elizaa62, 2019-02-02:
> Pozice děr v ose X jsou myslim vyřešené. Teď by chtělo ještě odstranit konstanty, které udávají vzdálenost kapes pro matky od stěny.
> Je to zřejmě poměrně složitá úprava stejně jako odlehčení, ale umožnilo by to měnit velikost podložky bez toho, aby se musely ručně posouvat díry na šrouby.

Řešilo se toto nějak? Moje paměť je totiž v koncích...

kaklik, 2019-02-02:
@Elizaa62 Já se na to všechno pokusím reagovat v jednom komentáři a aby to bylo trochu přehledné, tak to zkusím stručně v bodech. 

-  Nemáme teď už žádný platný výkres sestavy. Neplati proto žádný parametr, převzatý z výkresu a výkresové parametry je potřeba nahradit jinými parametry.  Důsledkem toho je, že bude zachován parametr určující pozici pilonu a ostatní parametry se z této pozice musí dopočítávat. Což asi může být i složité protože to bude záviset na rozměrech tištěného dílu/spojky na horní části vertikální trubky. 
- Šrouby v krytech jsou myslím vyřešené.
- Vzdálenost kapes pro matky od stěny ještě vyřešená není.

kaklik, 2019-08-13:
Tohle je zřejmě už neaktuální issue.

closed 2019-08-13 by kaklik