ThunderFly-aerospace/TF-G1 #66: Neznámé parametry v sestavě 666_2000

kaklik, 2018-01-26:
Pokusil jsem se vyrenderovat novou celkovou sestavu 666_2000.  Zjistil jsem, že tam je použita proměnná "okraj", která ale  zřejmě nikde není definovaná.

    openscad -m make -o STL/666_2000.stl src/666_2000.scad
    ECHO: 2.92971
    WARNING: Ignoring unknown variable 'okraj'.
    WARNING: Ignoring unknown variable 'okraj'.
    WARNING: Ignoring unknown variable 'okraj'. 
    WARNING: Ignoring unknown variable 'okraj'.
    WARNING: Ignoring unknown variable 'okraj'.

Elizaa62, 2018-01-27:
Tento problém jsem odstranila. Byl problém v tom, že jak se měnily názvy proměnných u té spojky 888_1001, tak v dílu 666_1027 byly špatné názvy. Akorát teď mám problém v tom, že mě ten model nefunguje.

Elizaa62, 2018-01-27:
Tys to nějak měnil? Mám tam ty osy naopak.. i díl 666_1027 je vložen špatně.

kaklik, 2018-01-27:
Ne osy jsem neměnil. Možná jsem akorát zase zapnul to nastavení pohledu v Openscadu.

kaklik, 2018-01-28:
Myslím, že to celé nefunguje správně díky tomu, že v souboru 666_1027 je použita proměnná "okraj", která ale nikde nemá nastavenou hodnotu:
[image]

Elizaa62, 2018-01-29:
Opraveno, ale stejně je ten soubor nějaký vadný.

kaklik, 2018-01-29:
Já nějak nevidím žádnou změnu. Možná jsi to sem nenahrála.

Elizaa62, 2018-01-29:
Já to sem píšu, abych na to nezapomněla. Nahraju to pak večer až pro dnešek skončím.

Elizaa62, 2018-01-30:
Nahráno a stále to nefunguje, jak má.

kaklik, 2018-01-30:
Já myslím že to funguje tak, jak je tam napsáno.   Pro vložení podložky 1027 je potřeba, aby tam byla správně nastavená rotace tj. 
	rotate([90,0,0])	
		666_1027();

místo  
	rotate([0,0,0])	
		666_1027(); 

A potom je ještě potřeba, aby podložka byla  ve svém souboru správně posunutá. Teď se po jejích úpravách stalo, že je moc nízko, takže to nefunguje správně.