ThunderFly-aerospace/TF-G1 #68: Výška 666_1027

Elizaa62, 2018-01-31:
Jsi si jistý, že ta podložka bude sedět na podvozkových trubkách?

kaklik, 2018-01-31:
Nejsem. :)

Elizaa62, 2018-01-31:
No podle toho, jak jsem tam dala ten spodní kryt, tak mi moc nepřipadá, že ta podložka bude sedět na těch trubkách a nebude se tam muset nic vykrajovat.

kaklik, 2018-01-31:
No,  a není to náhodou tím, že tu podložku máš ve špatné výšce?

Elizaa62, 2018-02-05:
Tak ta podložka je špatně posunutá.. Jenže teď je problém v tom, že ten horní kryt jsem udělala tak, aby to sedělo na tu podložku, tak jak tam byla.. teď to už zase nebude pasovat..

Elizaa62, 2018-02-05:
[image]

Elizaa62, 2018-02-05:
Nevím.. přijde mi, že v tom celkovém modelu nepasuje už vůbec nic.. :(

kaklik, 2018-02-06:
Zřejmě je potřeba, aby byla nejdříve správně umístěna podložka a to umístění bylo dáno společným parametrem přes všechny modely. 

Teprve potom se mohou uprovovat oba kryty tak, aby pasovaly na podložku.  A ty úpravy zas musí vycházet ze společných parametrů. 

Pokud se posunutím podložky rozbil ten model, tak je to špatně protože to znamená, že se hýbalo s parametrem o kterém ty ostatní modely neví.

kaklik, 2018-02-06:
Ještě mě napadlo, že hodně  těchto potíží s tím, že to nepasuje by zmizelo. Kdyby se na tu kapku udělal nový společný soubor.  A zrušila by se tak například varianta drop_B a ty divné posuny o nějaká čísla.  Místo toho by byl jeden společný model pro kapku ze kterého by se parametrem vyřezával spodní, nebo horní díl, podložka a podobně.  
Pak se totiž nemůže stát, že by na sebe ty části nenavazovaly. Neboť, když se navzájem neposunou, tak podložka bude vznikat ořezem a z jednoho modelu vyjde i jak horní tak i spodní kryt. Tím pádem pak musí nutně například spodní a horní kryt na sebe plynule navazovat a ten pak musí navazovat na podložku.  Pozice dělení pak můžou být dané spočítanými parametry  (třeba z tloušťky podložky).
Aby to takhle bylo, tak myslím ani není potřeba ty modely nějak zásadně předělávat. V podstatě v tom stačí uklidit a možná částečně vyřešit. #62

Elizaa62, 2018-02-10:
Výška vyřešena.

kaklik, 2018-02-10:
Souhlasím, může se to zavřít. :)

closed 2018-02-10 by Elizaa62