ThunderFly-aerospace/TF-G1 #70: Soubor 666_1025

Elizaa62, 2018-02-07:
Horní kryt opraven, základní kapka vytvořena v souboru 888_1000. Pokud bude vše správně fungovat pro tisk, tak se issue uzavře.

kaklik, 2018-02-09:
Zjistil jsem, že zámky na horním krytu jsou zřejmě delší než otvory v protikusu. Ten rozdíl je malý, asi něco jako 0.3mm. Nicméně to způsobuje, že části nejdou sesadit až k sobě a musí se zámky trochu upilovat. 
Všiml jsem si, že tahle potíž existuje už dříve. Ale teď je ten tisk už tak dobrý, že je to zřejmé. 

Řešením je zavedení tolerancí v rozměrech zámků. Klidně se na to může požít existující proměnná "global_clearance".

Elizaa62, 2018-02-10:
Ta proměnná je tam již použita. Tak já ji tam dám asi ještě jednou.

kaklik, 2018-02-10:
Jo, vypadá to, že jsi na to přišla.  :) Já jsem taky viděl, že ta tolerance tam už nějak je. Ale myslel jsem si, že asi není použitá správně, nebo ve správné ose.

Elizaa62, 2018-02-10:
takže zavírám :)

closed 2018-02-10 by Elizaa62