ThunderFly-aerospace/TF-G1 #74: Nastavení výšky krytu pylonu (díl 666_1032)

kaklik, 2018-02-11:
Líbilo by se mi, kdyby kryt pro pylon 666_1032 byl vyřešený tak, že se jeho výška dopočítá automaticky, z délky vertikální středové trubky (pylonu).

Pak by bylo možné uříznout definovaně dlouhou vertikální trubku a vytisknout kryt, který k ní bude pasovat.  Poměrně výrazně to pomůže pro jednodušší nastavení polohy těžiště.

Elizaa62, 2018-02-11:
Myslím si, že jsem to vyřešila.

Elizaa62, 2018-02-11:
Máš to i s komentem.

kaklik, 2018-02-13:
Zkoušel jsem to použít a je to skoro dobře :) 

[image]

Akorát to myslím nezohledňuje výšku držáku krytu a kryt je tak přibližně o výšku držáku kratší. 

[image]

Elizaa62, 2018-02-13:
Teď asi nerozumím, co mám přesně opravit..

kaklik, 2018-02-13:
Myslím, že k tomu může pomoc horní obrázek, který jsem sem přidal.  Je na něm vidět, že kryt pylonu má pod horní spojkou která se nasazuje na trubku vůli.  To je úplně v pořádku. 
Ale zároveň pod krytem dole, je mezi kapkovitým horním krytem a krytem pylonu mezera. Což znamená, že když by se kryt pylonu nasadil na držák, který je na horním krytu kapky, tak jak bude reálně nasazený.  Tak kryt pylonu bude moc krátký.
Mezi spojkou na konci vertikální trubky a krytem pylonu vznikne myslim si až zbytečně velká mezera. Která tam jinak nemusí být skoro žádná, až na tolerance tisku.

Elizaa62, 2018-02-13:
Je mi divné, že mezi držákem na horním krytu a krytem pylonu je taková mezera. Myslela jsem, že výška toho držáku na horním krytu vůbec s ničím nesouvisí.

Elizaa62, 2018-02-13:
A dělal jsi to v té sestavě 888_1005?

kaklik, 2018-02-13:
Výška držáku na horním krytu je myslím doplněním do kvádru, kterým se ořezává kapka.   Konečná pozice krytu pylonu by měla myslím být taková, že bude nasazením přes držák doléhat až na horní stěnu horní kapky. 
Dělal jsem to v sestavě 666_2000, ale to by mělo být jedno, protože ta výška krytu pylonu se skutečně mění a to je to, co jsem chtěl ověřit.

Elizaa62, 2018-02-14:
Výška vyřešena.

kaklik, 2018-02-16:
Ano, připadá mi, že to funguje.

kaklik, 2018-02-26:
Tak se mi zdá, že výpočet výšky krytu zanedbává posun vertikální trubky pylonu nad centrální trubku. 
Podle mě je v modelu 1017 tahle vzdálenost tvořena součtem width/2 + (- global_clearance*2), chtělo by o věřit v nějaké celkové sestavě, že ta délka trubky je vážně vypočítána správně. 

[image]

Přišel jsem na to tak, že v Parameters.scad mám délku trubky nastavenou asi o 3mm kratší, než je reálná změřená.  Kryt by tak měl mít vůli ty 3mm.  Reálně je ale přesně na těsno.

Elizaa62, 2018-02-27:
Délku vertikální trubky máš v Parameters nastavenou, tak jak jsi chtěl. Tzn. 195 mm. A tady o tom posunu jsem ti psala, že si nejsem jistá, že je to správně nastavené. 

Ale pokusím se to dát nějak dohromady.

kaklik, 2018-02-27:
Udělal jsem pull-request, kde jsem to nastavení délky změnil.  Ono je to skoro dobře.  Jenom ten model krytu nepočítá s tím, že trubka pylonu je trochu výše, díky tomu, že nezačíná úplně přesně na stěně té horizontální trubky.

Elizaa62, 2018-02-27:
Výška dílu 666_1017 je konstantní, v kódu je nastaveno 70 mm. Mezeru mezi trubkami jsem nastavila na coupling_wall_thickness. Tím pádem si vytiskni díl 666_1017 znovu a už by to mělo sedět.

kaklik, 2018-02-27:
Dobře, ještě jsem to upravil tak, aby mezera mezi trubkami byla podle existující proměnné "thickness_between_tubes", která je pro tenhle rozměr myslím vhodnější. 
Na to, abych to otestoval musím přetisknout jak spojku, tak kryt, takže se to dám do tiskového plánu.

closed 2018-02-27 by kaklik