ThunderFly-aerospace/TF-G1 #75: Díry pro připevnění podložky k hlavní trubce

Elizaa62, 2018-02-11:
Vytvořila jsem ty otvory pomocí union, kdy jsem připojila  k podložce kvádry v určité šířce. Nejsem si ale jistá, zda to bude fungovat přesně, jak požaduješ, že se tam opravdu přidá materiál. Je to vytvořené hned v začátku hned za sloupky pro GPS. Tak na to prosím koukni.

kaklik, 2018-02-11:
Může to být dobře a nemusí.  Záleží na tom, jak se pak provede to odlehčení.  Protože odlehčení je pak potřeba provést tak, aby se neodečítalo od těhle připojovacích/vyztužovacích kvádrů.

kaklik, 2018-04-13:
Není tohle už tohle nějaké zastaralé issue?

Elizaa62, 2018-04-16:
Je, ale stále není vyzkoušeno, zda to bude fungovat.

kaklik, 2018-04-16:
Tak já to vyzkouším, když budu mít podle čeho vyvrtat ty díry. :)

kaklik, 2018-05-20:
Zkoušel jsem přišroubovat podložku na hlavní podélnou trubku.  Díry jsem si provizorně vyznačil přes vytištěnou podložku. 
Nejdelší šrouby M3, které jsem sehnal jsou dlouhé 50mm a zjistil jsem, že nedosahují skrz podložku tak, aby na druhé straně šla přitáhnout matka. 

Je zřejmě potřeba upravit hloubku otvorů tak, aby 50mm šrouby stačily.

kaklik, 2018-05-29:
Tak jsem potřebnou délku šroubů pro připevnění podložky zkrátil na 50mm  a zavedl jsem nový globální parametr, který tuto délku šroubu definuje. 

poslední šroub v zadní části podložky jsem uměle zkrátil na 40mm, což je teď sice špatně, protože správně by šroub asi chtělo posunout tak, aby šla zachovat délka.  Ale nemáme čas na výrobu celého nového modelu včetně výměny hlavní trubky.

Elizaa62, 2018-06-26:
Je u tohoto issue  něco, co mám opravit?

kaklik, 2018-06-26:
Já jsem teď rozhodnutý postupně sestavit jeden další nový kus vírníku.  Z toho důvodu by zřejmě bylo vhodné opravit i tu poslední díru v podložce, ve které je teď kratší šroub. 
Oprava spočívá ve vhodném posunutí šroubu do širší části podložky.

closed 2018-09-11 by kaklik