ThunderFly-aerospace/TF-G1 #76: Nefunguje správně parametr pro zapuštění akumulátorů do podložky

kaklik, 2018-02-11:
Když se parametr "sink_of_accumulator" nastaví na 0 nebo skolo nula. Tak to udělá něco jako tohle:

[image]

Pokud se nastaví na -17, tak je akumulátor naopak skoro na povrchu podložky.

Elizaa62, 2018-02-12:
Opraveno :)

kaklik, 2018-02-12:
Ještě to možná způsobuje jinou chybu v tomhle modelu: 

[image]

Elizaa62, 2018-02-13:
Opraveno.

kaklik, 2018-02-14:
Dobře, ještě ověřím, že když se vytiskne podložka a držák akumulátoru, tak že vyjde správně výška.  Pokud to tak bude, tak to zavřu. :)

kaklik, 2018-02-16:
Tak výška akumulátoru mezi podložkou a držákem sedí. :)

closed 2018-02-16 by kaklik