ThunderFly-aerospace/TF-G1 #77: Tloušťka stěny spodního krytu zřejmě není všude stejná

kaklik, 2018-02-12:
Při nastavování tisku spodního krytu jsem zjistil, že v některých částech je vygenerovaný kryt děravý. Myslím, že je to způsobené tím, že stěna je v některých místech tak tenká, že ji nelze vytisknout. 

[image]

Elizaa62, 2018-02-12:
Trochu dost nechápu, jak je to možné...

kaklik, 2018-02-12:
Mám se pokusit tu příčinu najít? :)  Myslím, že to bude jedna z těchto dvou možností. 

1. Model je vygenerovaný příliš hrubě, protože není správně aplikována proměnná "draft"
2. Stěna je tenčí díky tomu, že se odečítá špatně posunutá kapka. V tomhle případě asi posunutá moc do spodní části modelu.

kaklik, 2018-02-12:
Tak zřejmě to bylo skutečně způsobené příliš hrubým vykreslením zakulaceného povrchu, když jsem opravil v modelu několik drobných chyb a správně použil proměnnou "draft", tak se zdá, že model je zpracovaný, jak má být: 

[image]

closed 2018-02-12 by kaklik