ThunderFly-aerospace/TF-G1 #78: Odlehčení podložky 666_1027

kaklik, 2018-02-12:
Asi už by se dalo vrátit k tomu odlehčení podložky. Protože teď už snad mohu mezitím zkoušet tisknout ostatní modely.

Elizaa62, 2018-03-25:
Co je tam to za díru pod GPS sloupkama?

kaklik, 2018-03-25:
Zřejmě myslíš, ty dva symetrické otvory na trubky, které jsou po celé délce.  Protáhne se tudy trubka, která slouží místo zámků při lepení a nebo jako podélná výztuha, když se tam nechá. 
Psal jsem o tom v #84

Elizaa62, 2018-03-26:
V tom případě se bude muset odstranit jeden připevňovací šroub od krytu nebo se posune ta díra.

kaklik, 2018-03-26:
To je zajímavé, v modelu podložky který jsem vytiskl, tet otvor pro matku a šroub takhle umístěný není.  Tady je to vidět: https://github.com/Elizaa62/Components/blob/master/STL/print_plate_1027D.stl

Připadalo mi to docela v pořádku, až na ten šroub pro spodní kryt.  Ten jsem si ale říkal, že půjde taky posunout někam jinam.

Elizaa62, 2018-03-31:
Pokusila jsem se připravit funkci pro vytvoření odlehčení.. tedy funkci pro výpočet počtu možných odlehčujících čtverců (C) při daných rozměrech.  Ale vznikly mi problémy:

- Chtěla jsem použít funkci "Floor", abych dostala zaokrouhlený počet, ale funkce mi nefunguje. Nevím, kde dělám chybu. 

- Počet čtverců jsem počítala na přeponě trojhúhelníka, jejíž délku jsem si vypočítala pomocí vzdálenosti v ose X (A) a vzdálenosti v ose Z (B). Výpočet je vidět v kódu. Zde jsem se chtěla zeptat, zda se ta délka přepony počítá vždy od počátku souřadnicového systému? A jak to případně změnit, když chci začátek přepony v určitém místě v ose X?

kaklik, 2018-03-31:
1.  Floor ti nefunguje, protože ta funkce se jmenuje floor, nikoli Floor.  Je to citlivé na velikosti písmen. 
Pak to začne fungovat: 

    Compiling design (CSG Tree generation)...
    ECHO: "C is", 20.7958
    ECHO: "Floored C is", 20
    ECHO: "vzdalenost_x is:", 47.680
    ECHO: "vzdalenost_y is:", 20.604

2. Možná úplně nerozumím otázce, protože nechápu co je A.   Ale připadá mi, že to počítáš zbytečně komplikovaně. Očekával bych, že si spočítáš úhlopříčku mezi odlehčujícími čtverci. A tou vydělíš šířku podložky danou 2x tím, co nyní máš označené jako B na řádku 76.
A z toho co ti vyjde uděláš floor.

Elizaa62, 2018-04-09:
Nějak jsem se snažila vytvořit to odlehčení, ale nefunguje mi to tak jak bych potřebovala. V ose x mi to funguje, ale v ose na ni kolmé nevím jak to udělat, aby tam byl počet odlehčujících čtverečků jaký se mi vypočítá pomocí floor(C).

kaklik, 2018-04-09:
Tady budu muset vymyslet co s tím...  Mezitím asi můžeš klidně řešit jiný úkol, jestli máš čas. Například to vylepšení šroubového spoje horního krytu by se teď docela hodilo, abych mohl přetisknout všechny kryty.

kaklik, 2018-04-19:
Vypadá to, že to nefunguje kvůli tomu, že hodnota C se po několika cyklech dostane mimo smysluplný rozsah. 

    ECHO: "C is", 4.75473
    ECHO: 4
    ECHO: "C is", 4.34487
    ECHO: 4
    ECHO: "C is", 2.66448
    ECHO: 2
    ECHO: "C is", -0.25
    ECHO: -1
    ECHO: "C is", undef
    ECHO: undef

Vycházím z  tohoto bloku kódu 
https://github.com/Elizaa62/Components/blob/750adca36db7c6889d1e17afac46d3a3844a6573/src/666_1027.scad#L629-L652

Elizaa62, 2019-02-02:
Jak to vypadá s touto úlohou?

kaklik, 2019-02-02:
Nic na ní neproběhlo.

kaklik, 2019-08-04:
Tohle je nahrazeno lepším řešením z issue #223

closed 2019-08-04 by kaklik