ThunderFly-aerospace/TF-G1 #81: Vyztužení dílů B a C spodního krytu 1029

kaklik, 2018-02-14:
Při vytištění dílu B spodního krytu jsem zjistil, že vrchní stěna, která má být rovná se vlní.  Na fotografii to není až tak dobře viditelné.  Ale pomohlo by, kdyby žebra, která jsou teď na bočních stěnách byla i na horní stěně. 

Dále také přepážka, která odděluje díl B od dílu A není moc dobře vytištěná. Je vidět že na fotografii prosvítá.  To je zase tím, že přepážka při tisku není ničím podepřená a tiskne se do vzduchu.  Dalo by se to vyřešit tím, že žebra z horní stěny budou pokračovat až přes přepážku na protější zaoblenou stěnu.   Tím se stane, že při tisku se budou nejdříve tisknout žebra a pak teprve přepážka, která ale už bude podepřena vytištěnými žebry.

[image]

Elizaa62, 2018-02-14:
Prozatím vyřešeno.

kaklik, 2018-02-18:
Vytiskl jsem si zkrácené varianty dílů, abych viděl, co udělá mřížka pod přepážkou. Stalo se tohle: 

[image]

Je to způsobené tím, že slicer při plánování tras tisku neplánoval tisk přepážek přes celou délku, ale jenom ve čtvercích. Takže se tahy vzájemně nespojily. 
Myslím, že jsem to vyřešil tím, že jsem odebral příčné výztuhy tak, aby tisk byl vždycky přes celý rozměr modelu.

[image]

Přepážky jsou sice trochu prověšené, ale konstrukční účel plní. :)

closed 2018-02-18 by kaklik