ThunderFly-aerospace/TF-G1 #82: Optimalizace dílů A a D spodního krytu pro tisk

kaklik, 2018-02-15:
Potíž s tiskem dílů, které mají otvory pro podvozkové nohy je taková, že pokud se díl tiskne v poloze "na lemu pro podložku" tj. takto: 

[image]

Tak  nedopadne dobře tisk kulaté části, která v takové  poloze tvoří převis. 

Pokud se díl otočí do této polohy 
[image]

Tak tisk z větší části dopadne dobře, až do chvíle než se začnou tisknout výřezy pro podvozkové nohy. Horní strana výřezu (v této poloze) se totiž zhroutí, protože není ničím podepřená. 

### Návrh na řešení 

- Doplnit výřez pro podvozkovou nohu tenkým čtvercem, o tloušťce stěny třeba 0.6mm.  Který bude sloužit k podepření horní strany výřezu během tisku. 
Tento čtverec se pak po vytištění modelu vystřihne, nebo vylomí. 
- Doplnit lem na spoj dílů C a D.   To z důvodu aby díl D při tisku lépe držel na podložce. 

V principu stejné úpravy se mohou udělat na dílu A.

Elizaa62, 2018-02-15:
[image]

Elizaa62, 2018-02-15:
Doplněn i lem.