ThunderFly-aerospace/TF-G1 #83: Tolerance bočních zámků na krytech

kaklik, 2018-02-15:
Na horním krytu jsem zjistil, že tolerance bočních zámků nejsou výškově asi úplně správně.  Viz následující fotografie. 

[image]

[image]

Horní zámek ale vychází v pořádku, tak jak bych čekal :) 

[image]

Konkrétně jsou tohle části D a E horního krytu.

Elizaa62, 2018-02-15:
Měla jsem špatná znaménka u tolerancí. Nevšimla jsem si, že se výřez pro zámek zvětšuje nahoru a posunula jsem ho o půlku tolerance nahoru místo dolů. Takže by to mělo fungovat.

kaklik, 2018-02-21:
Ano, potvrzuji, že je to vyřešené. :)

closed 2018-02-21 by kaklik