ThunderFly-aerospace/TF-G1 #84: Dělení podložky 1027

kaklik, 2018-02-15:
Při slepování vytištěných dílů podložky jsem přišel na tyhle záležitosti: 

- Pomohly by nějaké zámky, které zajistí, že díly se při slepování nemohou vzájemně posunout. Ani na výšku ani na šířku. 
- Pomohlo by, kdyby dělení podložky bylo nastavené tak, aby v dělící rovině nebyl žádný šroub, nebo nějaký jiný složitý tvar.

Elizaa62, 2018-02-15:
Dělení jsem změnila. Otvory pro připevnění k horizontální trubce jsem posunula. Přidala jsem zámky. Koukni se, zda to může takto býti.

kaklik, 2018-02-21:
Jo v principu to tak asi být může, jenom je ještě potřeba zjistit, jak se to snese s odlehčením podložky #78.
Zatím to ale zavřu.

kaklik, 2018-02-22:
Tak jsem zjistil, že podložku v současné podobě nedovedu vytisknout, protože v tiskové sestavě mi vznikne neřešitelný problém s pozicí posledního dílu.  Který stojí na vystouplých zámcích, taktéž díl před ním by potřeboval, kdyby zámky mohly být otočené. 

https://github.com/kaklik/Components/blob/master/STL/print_plate_1027.stl

Zřejmě je to potřeba vyřešit tak, že zámky budou buď vypadat tak, že na nich jde vytisknout celý díl. A nebo bude existovat spojovací kus, který se zalepí do dílu, který by nešel se zámkem vytisknout.

Elizaa62, 2018-03-05:
Dalo by se ty křížky vytisknout samostatně a pak by se vlepily do otvorů v podložce a pomocí nich by se ty díly podložky slepily k sobě?

Elizaa62, 2018-03-05:
Hotovo.

Elizaa62, 2018-03-05:
Existuje spojovací kus 888_1006, který se vlepí do připravených otvorů v dílu a pomocí druhého konce se spojí s druhým dílem.

kaklik, 2018-03-05:
Vyzkouším to. Má ten díl nějakou toleranci, aby do toho otvoru šel vlepit?

Elizaa62, 2018-03-06:
Díra má toleranci.

kaklik, 2018-03-06:
Tiskne se to. Tisk podložky trvá 39 hodin.

kaklik, 2018-03-08:
Tak výsledek není nic moc. 

[image]

Řekl bych, že těmi propojovacími křížky cesta nevede.  Je to náročné na kvalitu tisku. 

Navrhoval bych několik změn. 

1. Posunout dělení podložky tak, aby spodní strana tištěného dílu  byla vždycky plocha. 
2. Do podložky vytvořit podélné díry, které budou sloužit pro zasunutí vodícíchtrubek, jednak sloužících pro slepení a možná i pro vyztužení, pokud se v modelu nechají.

kaklik, 2018-03-08:
Myslím, že jsem to vyřešil.  Pro sesazení dílů během lepení se prozatím použije stejná trubka, jako na podvozek.  
A dělení podložky jsem posunul tak, aby bylo zarovnané se stěnami spojek.

commit c8746d0b (2018-03-13): Fixed  #84
Pousunul jsem dělení tak, aby při tisku nevznikal žádný most.
Zámky jsem nahradil podélnou trubkou na kterou se části podložky během lepení nasunou.
closed 2018-03-13 by Elizaa62