ThunderFly-aerospace/TF-G1 #87: Do šroubového spoje dílů B a C spodního krytu nejde nacpat šroub

kaklik, 2018-02-18:
Šroub se při pokusu o vložení do otvoru pro stažení dílů zkříží v díře oproti svojí hlavičce.

[image]

Zřejmě je okolo otvoru pro šroub příliš malá tolerance pro hlavičku.  

Nevím jestli je dobře, že zapuštění není šestihranné a tím pádem nebude držet matku, která je na druhé straně. Zatím mi připadá, že matku tak nepůjde utáhnout.

Elizaa62, 2018-02-20:
Ale to zapuštění jsem dělala šestihranné...

kaklik, 2018-02-21:
Zapuštění je teď opravdu šestihranné, ale stejně to nefunguje.   Je potřeba, aby i díra pro šroub měla stejný úhel jako zapuštění a otvor na matku aby byl řešen stejně, jako jsou kapsy na podložce.

Elizaa62, 2018-02-25:
Změnila jsem celý šroubový spoj. Díra je umístěna pod úhlem, na matku je vytvořena kapsa a celé je to umístěné v kvádru.
[image]

kaklik, 2018-02-27:
Tak zatím jsem vytiskl jednu polovinu do otvoru na matku se nevejde matka, protože otvor na matku je nižší, než je výška matky. 
Je možné, že je to tiskem, neboť kapsa má trošku zborcený strop. nicméně tam není na to žádná rezerva. 

Jsou rozměry kapsy stejné jako u podložky?

kaklik, 2018-02-27:
Ještě by bylo fajn, před tím než to budu tisknout přidat ten lem i na tu horní stěnu, jak jsme o tom mluvili.

Elizaa62, 2018-03-02:
Lem přidán, otvor prohlouben, otvor na matku zvětšen z každé strany o global_clearance.

kaklik, 2018-03-04:
Ta úprava kupodivu způsobila, že je šroub teď více utopený v modelu. Ale zatím nevadí, zkusím to vytisknout. 
To přidání 2*global_clearance na výšku matky M3 způsobilo, že je ten otvor náhodou stejně velký jako ty na podložce. Nevím ale, jestli by v tomhle případě nebylo lepší změnit hodnotu výšky matky.  A sjednotit to nastavení otvorů jak tady, tak i u podložky.

Elizaa62, 2018-03-05:
Proč bys chtěl měnit výšku matky? Výška matky je pevně daná v Parameters. Ten otvor je stejně velký, protože v podložce přidávám na výšku matky 1 mm a tady 2*global_clearance, což náhodou vychází stejně.

kaklik, 2018-03-05:
To vím, že je to náhodou stejné. Ale jenom zrovna při tomhle nastavení global_clearance, když se to změní, tak to tak už nebude. Proto by myslím bylo vhodné změnit výšku matky, aby se global_clearance využívala jenom na zarovnání nepřesností tisku, ne rozměrů matky.

Elizaa62, 2018-03-05:
Proměnná global_clearance je u otvorů pro matky vyrušená. Je tam daná pevná hodnota a to hodnota 1.

kaklik, 2018-03-06:
Tu hodnotu 1 je myslím potřeba všude smazat. A místo toho nastavit  Nut_height_M3 v Parameters.scad  na hodnotu 2.7, místo 1,7mm.

Elizaa62, 2018-03-14:
Nut_height_M3 zvýšen z 1,7 na 2,7 mm. Opraveno v Parameters i v podložce a spodním krytu. 

Dále u spodního krytu jsem změnila tu kostku pro ochranu spoje dílu 2 a 3 na spojení válce a kužele.

kaklik, 2018-03-15:
Je tam určitě commitnutá správná verze? Protože mě připadá, že se to nezlepšilo.

[image]

Elizaa62, 2018-03-15:
Máš pravdu.. v jedné části jsem to zapomněla opravit.

kaklik, 2018-03-22:
Tak jsem zjistil, že přidaný kužel ve kterém je šroubový spoj zapuštěný, nepřechází plynule do stěny, jak by se asi dalo očekávat.  Místo toho trčí mimo.

[image]

Myslím, že by se to dalo celé elegantně vyřešit tak, že by se sestava kužel - válec - kužel. Nahradila sestavou dvou šikmých válců skrz stěnu.  Není přece nutné, aby šroub byl palalelně se stěnou.

Elizaa62, 2018-03-26:
Pozice spojení zavedena podle funkce "surface_angle" a "surface_distance". Obal spojení je pomocí válce s koulí na konci. Otvor pro hlavičku šroubu je válec.

Takto je vytvořeno spojení mezi díly C a D u horního krytu a u dílu 2 a 3 u spodního krytu.

kaklik, 2018-03-26:
Ano s tímhle výsledkem jsem spokojený. Myslím že je to i docela pěkné :) 

[image]

closed 2018-03-31 by kaklik