ThunderFly-aerospace/TF-G1 #88: Krátké zámky na spoji spodního krytu dílů D a E

kaklik, 2018-02-19:
Zámky vedoucí do dílu E se mi zdají nějak neúměrně krátké oproti vykousnutí v dílu E. 

[image]

Elizaa62, 2018-02-20:
Toto by mělo být také opraveno.

kaklik, 2018-02-21:
Teď je to opravené tak, že tyhle zámky jsou kratší než všechny ostatní.  Považoval bych za správnější, kdyby všechny zámky byly stejně dlouhé. Klidně zas na společný parametr.

Elizaa62, 2018-02-26:
Zaveden nový parametr "lock_length", který definuje délku zámku. Od této délky se poté v kódu odečítá posunutí "hull_wall_thickness", aby bylo zajištění připojení k dílu.

kaklik, 2018-02-27:
Tohle tedy znamená, že to mám vytisknout? Já jsem totiž čekal, jestli náhodou stejným způsobem neupravíš i ostatní zámky. 
Ale asi to teď na těchto dvou dílech, mám vyzkoušet, že?

Elizaa62, 2018-02-27:
Vyzkoušej to prosím :)

kaklik, 2018-02-28:
Tak zámky pasují, ověřeno tiskem.

closed 2018-02-28 by kaklik