ThunderFly-aerospace/TF-G1 #89: Špatná pozice výřezu na podvozkovou nohu v dílu 1029 E

kaklik, 2018-02-19:
[image]

Výřez na podvozkovou nohu je ve spodním krytu zřejmě na špatném místě vůči podložce a spodní kryt tak díky tomu nejde úplně nasunout na podložku.

Elizaa62, 2018-02-19:
Jak je to možné? V tom celkovém modelu mi to všechno krásně sedí..

Elizaa62, 2018-02-19:
Nevím,co mám tady opravit, když mi to v tom modelu sedí..

kaklik, 2018-02-19:
Pokud jde o model 888_1005.scad, tak tam bych neřekl, že to dokonale sedí.  Také v modelu vidím, že nesedí díry na šrouby v krytu s dírami na šrouby v podložce. 

[image]

v modelu 666_2000 je tahle záležitost ještě markantnější, ale zase tam sedí díry na šrouby v podložce s dírami v krytu. 

[image]

kaklik, 2018-02-19:
Když už se v tom vrtám, tak podobná nesrovnalost je v mnohem menším i u předního krytu. 

Tady je ukázka z modelu 888_1005:

[image]

Podobně i sestava 666_2000:

[image]

Elizaa62, 2018-02-20:
Ty díry pro šrouby neseděly proto, protože jsem posunula horní i spodní kryt  v ose x o mínus hull_wall_thickness. Vymazala jsem to a teď by ti díry pro šrouby měly sedět.

Elizaa62, 2018-02-20:
Koukni na to, ale myslím si, že díry pro podvozkové nohy by už měly sedět. Chyba vznikla tím, že jsem v zadní části díru posunul navíc ještě o tloušťku stěny spojky, neuvědomila jsem si, že tam tento rozměr nepatří.

kaklik, 2018-02-21:
Ano, je to opravené. :)

kaklik, 2018-02-21:
Ještě jsem zjistil, že v přední části by zřejmě bylo potřeba opravit tenhle zub: 

[image]

A v zadní části taktéž opravit zub a posunout lem: 
[image]

Obecně by asi bylo fajn, kdyby se díra ve stěně a díra v lemu vytvářela najednou.

Elizaa62, 2018-02-25:
Zub je odstraněný. Zmenšila jsem výšku výřezu. 

Tvorba lemu je posunuta do horní části kódu, tudíž s měnící se pozicí otvoru pro podvozkovou nohu se bude měnit i otvor v lemu.

kaklik, 2018-02-28:
Ověřeno vytištěním.

closed 2018-02-28 by kaklik