ThunderFly-aerospace/TF-G1 #91: Vyztužení ocasních ploch 666_1028

kaklik, 2018-02-21:
Současná konstrukce ocasních ploch má příliš malou tuhost. Řešením je z dílu zrušit prořezy a nahradit je výztuhami v dutém prostoru profilu.  K tomu je potřeba díl nejdříve upravit jako dutý model.  Do vnitřního prostoru je pak potřeba vymodelovat výztuhy. 

Výsledek by pak bude vypadat podobně, jako tento řez křídlem: 
[image]

Elizaa62, 2018-03-04:
V   " //LOWER - od osy x do plus y" jsem udělala dutý profil a pokusila se vytvořit nějaké výztuhy. Koukni na to prosím... Jinak ostatní profily jsou vytvořeny zatím jen jako duté.

kaklik, 2018-03-04:
Ano, tak nějak jsem si to představoval. 

[image]

Akorát mi není jasné, proč se zas děje to, že ten profil nemá všude stejně tlustou stěnu. Ale řekl bych, že to bude nějak souviset s tím, že parametr naca musí být vždycky celé číslo, takže třeba: naca = 0005. A to na řádku 7 tak není.

Jinak ty moje pokusy o výztuhy jak podélné, tak i ty prořezy se můžou dát pryč. Nefungují dostatečně dobře a kazí to tyhle tvoje nové výztuhy.

Elizaa62, 2018-03-05:
Na tom řádku 7 to byl jen překlep, kterého jsem si nevšimla.

Elizaa62, 2018-03-05:
Výztuhy jsem doplnila i do ostatních profilů, v řádku 7 jsem překlep opravila. Tvé výztuhy jsem vyrušila.

kaklik, 2018-03-05:
Paráda :) 
Akorát mi teď došla ještě jedna věc a to, že ty nové výztuhy zřejmě nemůžou být při tisku vodorovně, protože by se všechny tiskly jako most a nedopadlo by to myslím dobře. Stačí to ale zřejmě vyřešit tak, že se natočí úhlopříčně.  Tady jsem se to pokusil načrtnout. 

[image]

Elizaa62, 2018-03-05:
Výztuhy vytvořeny úhlopříčně.

kaklik, 2018-03-06:
Úhlopříčné výztuhy jsou vytvořeny úplně správně. 

Pokusil jsem se to vytisknout, abych zjistil, jestli to půjde a skončil jsem na tom, že model má u náběžné hrany profilu tenké stěny, které jsou pod minimální tisknutelnou tloušťkou. Domnívám se, že to způsobuje výpočet nebo posunutí rozměru vnitřní dutiny. 

[image]

Příliš tenká stěna je i na bočních stěnách profilů.

kaklik, 2018-03-07:
Pokusil jsem se to opravit, ale zjistil jsem, že je potřeba vnitřní dutinu posunout nějak neúměrně moc daleko od náběžné hrany. Skoro to vypadá, že u takhle tenkých profilů nějak nefunguje ten použitý postup výpočtu zmenšené kapky.

kaklik, 2018-03-15:
Tak myslím, že jsem to opravil, jenom ještě musím doladit nastavení pro tisk. :)

closed 2018-03-15 by kaklik