ThunderFly-aerospace/TF-G1 #93: Otvory pro šrouby v zadní části horního krytu 1025

kaklik, 2018-02-24:
Díly D a E horního krytu nemají otvory na stejných pozicích, jako jsou otvory pro šrouby v podložce.

kaklik, 2018-02-26:
Podobný problém má i spodní kryt.  Zřejmě by bylo vhodné, aby se otvory na šrouby přebíraly z nějakého seznamu otvorů, podobně, jako  se to dělá s dělením.  Podrobněji jsem to už popisoval tady  #62.

Elizaa62, 2018-02-26:
Jak je možné, že tam jsou otvory pro šrouby navíc? Dávala jsem si veliký pozor, aby šrouby přesně seděly na otvory v podložce..

kaklik, 2018-02-26:
Já nevím jak se to stalo. Ale v tom modelu je to lepší než v realitě. V modelu vidím akorát dvě díry v krytu, které nemají díry v podložce.  Zřejmě mám vytištěné nekompatibilní verze krytů a podložky, takže to přetisknu.

Tady na obrázku je to ta nejvíc vpravo: 

[image]

z druhé strany krytu je symetricky ta druhá..

Pak jsem se díval ještě na podložku. A tam to vypadá takhle: 
[image]
Je zvláštní, že tam je ta slepá díra a kapsa na matku, která zasahuje do drážky pro lem.

Elizaa62, 2018-02-26:
Řekla bych, že to bude tím, že máš špatnou podložku. Zkus když tak vytisknout jeden zadní díl podložky a vyzkoušet to. A já bych to pak případně opravila :)

kaklik, 2018-02-26:
Ale ta jedna slepá díra je v tom modelu i teď. Takže to se přetištěním nezlepší.

Elizaa62, 2018-02-26:
Pokusím se to nějak opravit.

Elizaa62, 2018-03-02:
Šrouby od spodního a horního krytu jsem zavedla pomocí globálních proměnných. Ta díra na šroub navíc, co je vidět na obrázku výše, už by tam být neměla.

kaklik, 2018-03-04:
Díra navíc tam už skutečně není. 
Jenom se teď spojily kapsy na matky pro obě matky do jedné, takže se myslím, že horní matka bude spadávat do díry pro spodní matku. 

[image]

Elizaa62, 2018-03-05:
Díru od spodního šroubu jsem posunula.

kaklik, 2018-03-06:
Souhlasím vyřešené.

closed 2018-03-06 by kaklik