ThunderFly-aerospace/TF-G1 #94: Spojení dílů C a D horního krytu 1025

kaklik, 2018-02-24:
Díly C a D horního krytu nemohou mít zámky. Mají pouze společný široký lem, kterým pasují k sobě. 
Při spuštění motoru se pod díl D ale dostává vzduch, který kryt strhává. Bylo by proto potřeba oba díly spojit. 
Řešením může asi být stejný šroubový spoj, jako je na nákladovém prostoru spodního krytu.

Elizaa62, 2018-02-26:
Až vyzkoušíš šroubový spoj u spodního krytu a bude jisté, že funguje, udělám ho i tady.

Elizaa62, 2018-03-05:
Přidala jsem stejné šroubové spojení jako je pro díl 2 a 3 u spodního krytu.

kaklik, 2018-03-06:
Vidím, že ty přidané šroubové spoje jsou z boku.  Myslím že v tomhle případě je potřeba je mít v té horní zakulacené části.  Protože hlavním účelem šroubů, je zajistit, aby byly díly C a D spojené v místě, kde je teď nic nedrží. To znamená poblíž držáku krytu pylonu.

Elizaa62, 2018-03-13:
Vytvořeno.

kaklik, 2018-03-13:
Je tam zkopírovaná ta chyba ze spodního krytu, kde je špatná výška matky v díře stejně jako v případě issue #87

[image]

Obávám se také že to provedení zapuštění šroubu, které tam teď je neumožní šroub dobře utáhnout, protože tam nepůjde strčit imbus v ose šroubu, ale bude muset být nakloněný ven.

Elizaa62, 2018-03-15:
Tak jsem se o něco pokusila, ale pochybuju, že ten můj způsob bude fungovat. Otvor pro zasunutí šroubu jsem zakončila pod větším úhlem.

kaklik, 2018-03-19:
Tak zatím to dopadlo tak, že ta díra pro hlavičku šroubu je moc malá.  
[image]

A asi bych měl nějaké nápady, jak to udělat celé lépe.  Zřejmě to ale bude potřebovat, abych připravil funkci, která vrátí úhel povrchu krytu v daném místě kapky.

kaklik, 2018-03-20:
Tak už existuje funkce, která ti umožní zajistit, aby díra pro šroub byla pod definovaným úhlem vůči povrchu krytu.

Elizaa62, 2018-03-21:
Díra pro šroub opravena funkcí "surface_angle".

kaklik, 2018-03-21:
Mě by se ještě líbilo, kdyby ten válec ve kterém jsou zapuštěné šrouby z vnitřní strany postupně splýval s tím krytem. Takhle jak je tam ta špička, tak mi to připadá hodně divné.  Mě se docela líbilo to první řešení, kde byl šikmý válec skrz stěnu a postupně mizel do ztracena.  Pro šroub je to výhodné, protože válec pak může mít osu shodnou s osou šroubu. Na straně s matkou pak bude akorát díra na matku šikmo.

kaklik, 2018-03-31:
Fajn, zatím tam necháme tyhle válce zakončené koulí. :)

closed 2018-03-31 by kaklik