ThunderFly-aerospace/TF-G1 #95: Rozdělení krytu 1032 na spodní a horní část

kaklik, 2018-02-24:
Přišel jsem na to, že by asi bylo praktické kryt pylonu rozdělit na horní a spodní kryt, které by byly vzájemně buď slepené a nebo sešroubované.  To by už záleželo na tom, jestli by sešroubování zároveň nějak i zajišťovalo pozici krytu. 
Každopádně takové řešení může mít tyto výhody: 

- Bude možné vytisknout kryt, který je vyšší než maximální tiskový rozměr tiskárny. 
- Tisk by mohl být kvalitnější, neboť nyní kryt nemá žádnou rovnou plochu, na kterou lze model při tisku postavit a celý kryt se tak musí tisknout na podporách
- Rozdělení krytu, by mohlo umožnit zlepšení uchycení krytu k pylonu.

Elizaa62, 2018-02-25:
Takže bychom díl rozdělili na dva a ke každému bychom přidali lem pro slepení?

kaklik, 2018-02-25:
Ano, to by mohlo být dobré řešení. Slepený lem ten kryt rovnou i vyztuží, takže bude pevnější.

Elizaa62, 2018-02-25:
Kubí, jak tento díl tiskneš? Délka dílu 666_1032 v ose X je větší než 150 mm..

kaklik, 2018-02-25:
Úhlopříčně na podložku 150x150mm se to ještě vejde. :)

Elizaa62, 2018-02-25:
Vyřešeno. Doplněn lem a do parameters.scad přidán další globální parametr "cover_pilon_position" pro dělení na výšku, tzn. v ose Z.

kaklik, 2018-02-26:
Ověřeno vytištěním modelu a díly pasují.

closed 2018-02-26 by kaklik