ThunderFly-aerospace/TF-G1 #98: Šířka podložky 1027

kaklik, 2018-02-27:
Zjistil jsem, že šířka podložky byla nastavená přesně na šířku trupu tj. 150mm. Jenomže má pasovat dovnitř horního krytu. To znamená, že musí být menší než kryty a to ještě navíc o toleranci. 
Podložku jsem zmenšil a přidal tolerance.  Jenže se stalo, že z bočních stěn začaly vylézat ještě nějaké další přidané hranoly, nevím kde v kódu jsou. 

[image]

Elizaa62, 2018-03-02:
Vyřešeno.

kaklik, 2018-03-04:
Je fajn, že už tam ty výstupky nejsou. 
Nedovedu to ale ověřit tiskem, protože podložka nejde vytisknout díky současnému řešení zámků. Viz #84

Elizaa62, 2018-03-05:
Výstupky nejsou a řešení zámků je též vyřešeno, viz #84 .

kaklik, 2018-03-06:
Souhlasím, je to vyřešené.

closed 2018-03-06 by kaklik