ThunderFly-aerospace/TF-G1 #101: Svrtávací přípravek pro vertikální trubku

kaklik, 2018-03-03:
Ke spojce na vertikální trubku pylonu by byl potřeba přípravek na předvrtání děr do trubky.  Mělo by to fungovat tak, že trubku uřízneme na potřebný rozměr, začistíme řezy. 
A na oba konce se nasadí vytištěné díly, srovnají se na rovné podložce, tak aby byly stejně natočené.  A pak se na stojanové vrtačce provrtají všechny díry. 

Podobný přípravek jsem už udělal pro  díl 666_1006 a 666_1028. Takže z nich je možné vyjít.
Pro díly 666_1017 a 666_1026 přípravky chybí. 

Důležitou vlastností přípravku je, aby jeho rozměry byly svázány s parametry rozměrů modelu, který pak na vyvrtané otvory má reálně pasovat. A taky aby když se na oba konce trubky nasadí přípravky, tak aby šly srovnat rovnou podložkou. To znamená, že strana, kterou se budou pokládat společně na podložku ke srovnání musí mít stejnou tloušťku. Aby trubka byla na podložce rovně.
Pak jsem vyzkoušel, že stačí oba konce srovnat mírným poklepáním gumovým kladívkem, a přípravky se protočí do stejné roviny.

Elizaa62, 2018-03-09:
Tak jsem něco vytvořila. Je to díl 888_1007.scad.

kaklik, 2018-03-09:
Nějak jsem vůbec nečekal, že to uděláš jako samostatný nový soubor. Mě to připadá dost nepraktické :)  Protože bych čekal, že díly a přípravky k nim budou u sebe.
Ale vypadá to tak, jak jsem chtěl.  Akorát jsi nejsem jistý, jestli přípravek pro spojku 666_1026 bude pasovat k přípravku pro díl 666_1028. Když se budou vrtat díry do hlavní horizontální trubky a každý bude na jedné straně.

Elizaa62, 2018-03-12:
Jednotlivé přípravky jsem přiřadila k souborům, ke kterým patří. Soubor 888_1007 jsem použila pro vytvoření sestavy trubek a přípravků pro vrtání.

kaklik, 2018-03-19:
Teď mi došlo, že ten svrtávací přípravek pro díl 666_1028 nepasuje k těm novým přípravkům a asi by ho chtělo upravit. 
V celkové sestavě nyní totiž má otvory nahoru 

[image]

Ale vrtací přípravek pro díl 666_1026, který je vpředu má otvory horizontálně.   Správně mají ale ocasní plochy otvory ve stejné rovině, jako jsou otvory pro přední spojku.

Elizaa62, 2018-03-21:
Vrtací přípravek pro díl 666_1028 jsem opravila.

closed 2018-03-22 by kaklik