ThunderFly-aerospace/TF-G1 #104: Tolerance zámků na krytech

kaklik, 2018-03-13:
Vypadá to, že na krytech je nějaký zmatek v tolerancích zámků v ose X.  Zřejmě je to způsobené tím, že  mezi díly C a D se mění orientace zámků na krytech.  Proto nejde všechny zámky posunout o toleranci k jedné straně. 
Momentální situace je taková, že přední zámky nemají žádnou toleranci.  A zadní zřejmě správnou.

Elizaa62, 2018-03-13:
Horní kryt už také používá pro délku zámků parametr "lock_length".  

U předních zámků jsem opravila tu toleranci. Tak se prosím koukni, zda to je správně. 

Poté to dělení s "global_clearance/100".. u horního krytu jsem to dala místo "global_clearance/2". Před tím to "global_clearance/2" jsem ani nevěděla, že to tam je. Nejspíše jsi to tam vkládal ty a já si nevšimla :)

kaklik, 2018-03-14:
Jsou  ty tolerance určitě opravené dobře?  Protože mě se nezdá to, že jak u předních, tak i u zadních zámků krytu je teď přidaná tolerance  + global_clearance/2. 
Myslím si ale, že díky tomu, že mezi díly C a D se zámky otáčí, tak se musí změnit i znaménko tolerance, ne?

kaklik, 2018-03-14:
Tak už asi vidím, jak ty tolerance jsou vyřešené. 
Jenom mi ještě není jasné, jestli  přidání + global_clearance/2 na tomhle řádku je záměr?
https://github.com/Elizaa62/Components/blob/7b8f4cc0fb2d71bdaa4c620749f6f2d9c76078c9/src/666_1025.scad#L395

Je mi to podezřelé proto, protože na tomhle řádku pro dělení následujících dílů to naopak už není: 
https://github.com/Elizaa62/Components/blob/7b8f4cc0fb2d71bdaa4c620749f6f2d9c76078c9/src/666_1025.scad#L456

A myslím si, že je původní důvod přidání téhle tolerance to, aby na dělení dílů nezůstávaly zbytky lemů.  To by ale teď mělo řešit to zmenšení dílu o "global_clearance/100".

Elizaa62, 2018-03-14:
Já si nevzpomínám, že bych tam tu toleranci do řádku 395 přidávala. Protože je mi jasné, že to odstraňuje ten zbytek lemu, ale vážně si neuvědomuji, že bych to tam dávala. Každopádně v tom translate se to odstraní a ponechá se tolerance global_clearance/100 pouze u toho cube. Bude to tak správně?

kaklik, 2018-03-14:
Paráda, myslím že je to snad vyřešené. :)

Elizaa62, 2018-03-14:
Nahráno :)

Elizaa62, 2018-03-22:
Tak ty zámky by měly být vytvořeny už správně.

kaklik, 2018-03-22:
Ano to asi jsou.  Ale mě se nechce znova tisknout kryty, když na nich je potřeba opravit i ty další issues.

kaklik, 2018-03-31:
Zatím to zavírám, pravděpodobně je to vyřešené a seznam bude přehlednější.

closed 2018-03-31 by kaklik