ThunderFly-aerospace/TF-G1 #106: Vytvořit funkci která umožní získat úhel a vzdálenost povrchu aerodynamického profilu

kaklik, 2018-03-20:
Pro vyřešení úkolů  #94, #87, #78, #62 a možná #26  je potřeba vědět vzdálenost povrchu a úhel kapky v definovaném bodě. Aby bylo možné na povrch přesně umisťovat otvory a další elementy.

kaklik, 2018-03-20:
upravil jsem knihovnu naca4.scad tak aby poskytovala funkce 
-"surface_distance", pro zjištění vzdálenosti povrchu profilu od jeho osy 
- "surface_angle", pro zjištění úhlu povrchu vůči ose profilu. 

Tyto dvě nové funkce jsem vyzkoušel  tak, že jsem vytvořil příklad https://github.com/kaklik/Components/blob/master/src/naca4_test.scad

Který poskytuje takovýto výsledek.

[image]

Nyní je potřeba to implementovat do ostatních částí modelů.

Současná fuknce "trailing_edge_angle" se tak může zrušit.

Elizaa62, 2018-03-21:
Chápu to teda dobře tak, že všude v modelech, kde se vyskytuje funkce "trailing_edge_angle" ji mám nahradit funkcí "surface_angle"?

Elizaa62, 2018-03-21:
Pokusila jsem se do horního krytu vložit funkci "surface_angle", ale celý ten model se mi rozbil. Nejprve v dílu 666_1025 zmizel kus držáku na kryt pylonu  a když jsem provedla změnu i v dílu 888_1000, tak se mi rozbila celá kapka.

kaklik, 2018-03-21:
Na díly s dutým profilem, jako je třeba držák krytu pylonu jsem už před tím vytvořil novou funkci, která generuje duté profily hollow_airfoil.  Kromě toho, že ti vytvoří profil rovnou o požadované tloušťce stěny, tak je i o něco rychlejší.  

Zkusim to opravit a lépe okomentovat.

closed 2018-03-22 by kaklik