ThunderFly-aerospace/TF-G1 #108: Použití nových funkcí surface_angle a surface_distance

kaklik, 2018-03-22:
ta nová funkce surface_angle je použitá špatně minimálně u modelu 666_1025.
Neboť funkce obsahuje parametr x, kterým se zadává v jakém místě po délce profilu má být spočítaný úhel. Rozsah tohoto parametru je 0 až 1.  Kde 0 je náběžná hrana profilu a 1 je odtoková hrana profilu.
V modelu 666_1025 ale tento parametr zůstal nastavený na konstantní hodnotu 0,6 z mého příkladu.  Správný postup je, že tento parametr se bude měnit podle požadované pozice šroubu.

Elizaa62, 2018-03-22:
V tom případě tam je asi problém protože.. První obrázek je při X  = 0
[image]

A tento druhý obrázek je při X = 1
[image]

kaklik, 2018-03-22:
A když zkusíš nějaké jiné, než tyhle extrémní hodnoty?  Třeba 0.01 a 0.99 ? 
Je pravda, že jsem netestoval co se stane, když se tam dosadí meze. 
Je nahraný ten aktuální soubor, na kterém to testuješ, abych si to zkusil?

Elizaa62, 2018-03-22:
Je nahrane vse.

kaklik, 2018-03-22:
Myslím, že to funguje korektně i s mezními hodnotami

[image]

Chyba je v tom, že to nejdřív musíš správně otočit a pak teprve posunout. Jinak se ti otáčí okolo počátku souřadnic.

Elizaa62, 2018-03-22:
Opraveno.

kaklik, 2018-03-22:
Zřejmě to sem ještě není nahrané. 

Navíc je potřeba hlavně pomocí těchto funkcí umístit otvory pro šrouby v podložce. Protože tam to konečně zajistí aby všechny šrouby byly kolmo na stěnu a matky stejně hluboko pod krytem.

Elizaa62, 2018-03-23:
Neco jsem nahravala, ale toto to asi jeste nebylo. Jen jsem ti chtela napsat, ze se mi to podarilo opravit. Jeste jsem tam chtela neco dodelat, tak to nahraju pak dnes v noci :)

Elizaa62, 2018-03-23:
Tak jsem to nahrála. Otvory pro šrouby jsem vytvořila funkcí "surface_angle". Ale otvory byly těsně pod povrchem, tak aby mohly být trochu níž, tak jsem rozšířila na výšku otvor pro hlavičku šroubu. 

Jen je tam ještě detail, který je potřeba dodělat a to, že je třeba trochu pootočit otvor na matku, aby na špičky matky pěkně dosedal ten kvádřík, o který je otvor rozšířen na výšku.

kaklik, 2018-03-24:
Dobrá, ale pořád je to jenom kryt. V podložce 1027 žádné změny nejsou.

Elizaa62, 2018-03-25:
Změnila jsem pozice šroubů podle proměnné surface_angle. Pro každý typ šroubového spojení je zaveden globální parametr. Akorát jsem si všimla, že nefunguje u spodního krytu šrouby pro připevnění k podložce v části E. Hledala jsem příčinu a zjistila jsem, že u "bottom_screw_position" je smazaná pozice 5, která byla určena pro pozici šroubů v části E. Tys ji mazal?

Elizaa62, 2018-03-25:
V krytu jsi to právě nechal. Proto mě to zmátlo.

kaklik, 2018-03-26:
Tak jsem to tam nechal omylem. Promiň.

Elizaa62, 2018-03-26:
Změna šroubů v podložce i v krytech podle "surface_angle" a "surface_distance".

kaklik, 2018-03-26:
Muselo ti dát hroznou práci tam všude napočítat ty přesné parametry jako třeba:
hull_drop_length * 0.0927866

Myslel jsem, že to použiješ tak, že místo toho čísla tam přímo vložíš tu funkci. Takže třeba tohle:
```
translate([hull_drop_length * (bottom_screw_position[4]/hull_drop_length), main_tube_outer_diameter/4, - hull_drop_length * 0.0927866])
                rotate([0, -17.0902,0])   

```
Upravíš  nějak takhle: 
```
translate([hull_drop_length * (bottom_screw_position[4]/hull_drop_length), main_tube_outer_diameter/4, - hull_drop_length * surface_distance(x = bottom_screw_position[4]/hull_drop_length , naca=hull_airfoil_thickness, open = false)])
                rotate([0, surface_angle(x = bottom_screw_position[4]/hull_drop_length, naca=hull_airfoil_thickness, open = false),0])   

```

Může to tak být proto, protože funkce se chovají stejně, jako proměnná.  Proměnné navíc jdou přepisovat. Takže jiná možnost je mít u každého šroubu spočítané hodnoty v proměnných třeba screw_angle a screw_distance.  U dalšího šroubu se tyhle proměnné zas přepíší na aktuální hodnotu funkcemi a tak dále.  Ten model se počítačem čte postupně. Takže jeden název proměnné lze použít víckrát.

Elizaa62, 2018-03-28:
Já tam ten kód takhle vypsala, ale nefunguje to správně a já opět nevím, kde je chyba... :(
[image]

kaklik, 2018-03-28:
Připadá mi, že jsem to opravil. 

[image]

Hlavní chyba byla v tom, že jsi tam měla použitou proměnnou "hull_wall_thickness" místo "hull_airfoil_thickness". 
Vytvořím na tu opravu zas pull-request.

Elizaa62, 2018-03-30:
Snažila jsem se to opravit, ale ty spojovací šrouby se mi nějak pokazily.. Zítra to opravím a nahraju novou verzi.

Elizaa62, 2018-03-31:
Opraveno a nahráno.

kaklik, 2018-03-31:
Ano, teď to vypadá, že u horního krytu to už funguje správně.
Díval jsem se na model podložky a spodního krytu a připadá mi, že tam je to umístění šroubů zatím řešené tak, že se míchá  starý a nový způsob.  Je nějaký důvod, proč nový způsob s použitím funkcí nejde někde použít?

Elizaa62, 2018-04-03:
Tam, kde nebyl žádný úhel, jsem nechala ten starý kód pro posunutí, protože mi přišlo zbytečné to opravovat, když je tam rovná hrana. Ale už jsem tam to opravila a do posunutí dala ten nový způsob.

closed 2018-04-19 by kaklik