ThunderFly-aerospace/TF-G1 #109: Zajištění vrtacích přípravků proti otočení

kaklik, 2018-03-22:
Zjistil jsem, že je potíž zajistit, aby se během vrtání přípravek neotočil, ale přitom šel nasadit na trubku.  

Napadá mě to vyřešit tak, že by kostka přípravku byla třeba do 2/3 proříznutá a na kraji by byl šroub, kterým by kostka šla stáhnout ve vyrovnané pozici na stole. 
Takže by se postupovalo tak, že se přípravky nasadí na trubku, na stole se vyrovnají. Utáhnou se zajišťovací šrouby. Na vrtačce se vyvrtají díry, povolí se zajišťovací šrouby a sundají přípravky.

Elizaa62, 2018-03-26:
U dílu vrtacího přípravku pro díl 666_1026 vytvořen prořez a otvor pro zajišťovací šroub.

Elizaa62, 2018-03-27:
Máš to vytvořené u přípravku k dílu 666_1028. Snad jsem to pochopila správně.

kaklik, 2018-03-27:
Pochopila jsi to úplně správně. :)

kaklik, 2018-03-27:
Teď jsem tam ještě opravil to, aby šroub nezasahoval do trubky.  A aby prořez začínal až těsně u stěny trubky - to kvůli tomu, aby se nebortilo dno přípravku při stahování šroubem.

Předchozí stav
[image]

Po úpravě: 
[image]

Elizaa62, 2018-03-28:
Koukala jsem na ten změněný přípravek a všimla jsem si, že jsi šířku toho prořezu změnil
[image]

Mám to změnit i u těch ostatních přípravků?

kaklik, 2018-03-28:
Ano, nastavil jsem to podle global_clearance, protože mi to připadá logičtější. Šířka prořezu by totiž při ideální přesnosti výroby mohla být nulová. Taková situace ale nenastane, protože základní tolerance výroby odpovídá parametru global_clearance. 
Šířku prořezu jsem tak nastavil na trojnásobek základní tolerance podle [pravidla 3-sigma](https://cs.wikipedia.org/wiki/Pravidlo_t%C5%99%C3%AD_sigma).

Myslím si, že správné je to takto nastavit i v ostatních přípravcích.

Elizaa62, 2018-03-31:
Tys sice nastavil šířku 3*global_clearance, ale už jsi to neposunul doprostřed. Tak jsem to opravila a šířka prořezů a je opravena i u ostatních vrtacích přípravků.

kaklik, 2018-03-31:
Ano, to je chyba, že jsem to neudělal. Je super, že jsi si toho všimla a opravila jsi to. :) 
Myslím, že ty přípravky, teď vypadají tak, jak jsem si představoval, že by to mohlo fungovat. Takže to teď zavřu.

closed 2018-03-31 by kaklik