ThunderFly-aerospace/TF-G1 #110: Vymodelování podvozkových ližin

kaklik, 2018-03-22:
Bylo by užitečné vymodelovat i podvozkové ližiny podle výkresu 666_1033.  Jsou k tomu tyto důvody

- Do dokumentace budeme potřebovat celkovou sestavu
- Model podvozkových ližin by umožnil vytvoření ohybacího přípravku.

Elizaa62, 2018-03-28:
Ližiny vytvořeny v dílu 888_1008. Díl 666_1033 jsem ponechala jako vzorový.

Ližiny jsem doplnila i do celkové sestavy do souboru 888_1005. Také jsem rovnou vložila díly pro připevnění koleček.

PS.: Omlouvám se za tvrdé y ve slově lyžiny.. až po napsání komentů jsem zjistila, že se to píše s měkkým i...

kaklik, 2018-03-29:
Paráda. Ten celkový model hned vypadá kompletněji. :)
Ten pravopis slova ližina má celkem [komplikovaný výklad](https://www.pravopisne.cz/2010/10/liziny-%C3%97-lyziny/). A ani si nejsem jistý, jestli jsem to vážně všude psal konzistentně. 

Ale ještě jsem v modelu ližin přišel na tyhle záležitosti:

- Začátek a konec trubky přijde zastrčit do dílu 666_1006. Na ližině je proto potřeba nechat rovný začátek a konec minimálně v délce tohoto dílu.
- Bylo by myslím lepší ližiny navrhnout jako dva díly.  To znamená třeba 888_1008_A a 888_1008_B. Pořadí se u letadel označuje z pohledu pilota, který by řídil a začíná se nejvíc zleva. V případě podélného dělení tak od předu.

Elizaa62, 2018-03-31:
Trubka prodloužena o výšku dílu 666_1006, která je 45 mm. 

Podvozkové ližiny jsou rozděleny na dva díly - 888_1008_A a 888_1008_B. Oba díly jsem vložila do sestavy 888_1005.

kaklik, 2018-03-31:
Ano, tohle jsem chtěl. 
Jenom jsem čekal, že oba díly budou v souboru 888_1008.scad  a v něm budou dva moduly 888_1008_A() a 888_1008_B()

Aby to zachovávalo stejný způsob dělení, jako už máme u ostatních dílů.

Elizaa62, 2018-04-02:
Tak je to vložené do jednoho dílu 888_1008.scad.

kaklik, 2018-04-11:
Tohle je taky myslím hotové.  Teď na to navazuje ten ohýbací přípravek, který nemám úplně detailně rozmyšlený.

closed 2018-04-11 by kaklik