ThunderFly-aerospace/TF-G1 #115: Otočení šroubů pro připevnění akumulátorů

kaklik, 2018-04-07:
Při montáži akumulátorů jsem zjistil, že se na šroub docela těžko nasazuje matka, když šroub vede zhora dolu. 
Když se šroub otočí, tak zas nedrží v díře dokud se na něj matka nenasadí.
Lepší by proto bylo, kdyby šroub byl v podložce zdola nahoru a byl chycený matkou, podobně jako na následující fotografii. 

[image]

Akorát matka by byla zapuštěná v podložce. Pak by se na šrouby nasadil třmen 888_1001 a přitáhl se další matkou.  Zapuštění pro matku na patce třmenu je proto potřeba zrušit, aby se matkou dalo pro dotažení otáčet.

Elizaa62, 2018-04-10:
Vytvořeno.

kaklik, 2018-04-11:
Ano, takhle jsem si to představoval.  Jenom mi připadá, že ta horní matka je zapuštěná nějak moc hluboko. Čekal jsem, že bude akorát tak, aby nepřečnívala přes povrch. Díval jsem se do kódu, jestli tam uvidím něco jako Nut_height_M3,  ale nic takového jsem tam nenašel. Takže to nedovedu sám rychle opravit.

Elizaa62, 2018-04-11:
Je to zadané posunutím v ose Y o 1.5*main_tube_diameter. Musíš to v tom případě posunout nahoru o tloušťku podložky a odečteš od toho výšku matky asi ještě s nějakou tolerancí, aby bylo zaručeno, že ta matka nebude vyčnívat.

Elizaa62, 2018-04-12:
Opravila jsem to.

Elizaa62, 2018-04-12:
výška dílu 666_1027

[image]

kaklik, 2018-04-13:
Vypadá to jako vyřešené.

closed 2018-04-13 by kaklik