ThunderFly-aerospace/TF-G1 #117: Vyztužení dílu A spodního krytu

kaklik, 2018-04-11:
Již před časem jsem do horního krytu umístil výztuhu, která zlepšuje tisknutelnost viz obrázek: 
[image]

Bylo by zřejmě potřeba podobnou výztuhu dát i do spodního krytu, aby se při tisku nebortil.  V případě spodního krytu je pravděpodobné, že by tam těchto výztuh mělo být více než jen dvě.

Elizaa62, 2018-04-12:
Výztuhy vytvořeny.

kaklik, 2018-04-13:
Jsou tam drobné chyby, které brání tomu, aby mělo smysl to zkušebně tisknout.

- [ ] Potřeboval bych, aby výztuha končila přesně v rohu modelu.  To znamená, aby podepírala model při tisku až do poslední chvíle. (do nejvyššího bodu)
- [ ]  Výztuha by měla končit přesně vlemu a nepřesahovat jej. 

[image]

- [ ]  Také je třeba odstranit přesah do vybrání pro podvozkovou trubku
[image]

Elizaa62, 2018-04-16:
Opravila jsem chyby a trochu změnila pozice ostatních výztuh.

kaklik, 2018-04-17:
Zkusím to vytisknout.

kaklik, 2018-04-18:
Tak výztuhy kvalitě tisku pomohly. A model je teď i výrazně pevnější.

[image]

Nějak se ale rozbila ta tisková podpora u podvozkové nohy.  Snažil jsem se to opravovat, protože na začátku to nefungovalo vůbec.  Teď to ale taky není úplně ideální nad otvorem pro nohu přibyl nějaký materiál, který tam být nemá. 

[image]

Díky tomu tisk na téhle straně nedopadne až tak dobře, jak by mohl.

Elizaa62, 2018-06-26:
Chyba vznikla tím, že čtverec pro přidání výztuh byl veliký místo main_tube_outer_diameter 2*main_tube_outer_diameter. Chybu jsem odstranila.

Je u tohohle dílu ještě nějaká chyba, kterou se mám zabývat?

kaklik, 2018-06-26:
Zkusím to vytisknout a dám vědět. Ale v tuto chvíli si myslím, že není. :)

kaklik, 2018-06-27:
Tak myslím, že to zafungovalo parádně. Díl se vytiskl mnohem lépe než minule. :) 

[image]

commit a592be2b (2019-07-04): resolved #117
closed 2018-06-27 by kaklik