ThunderFly-aerospace/TF-G1 #118: Svrtávací přípravek pro otvory k připevnění podložky

kaklik, 2018-04-13:
Pro vyvrtání šroubů chybí přípravek, který by to umožňoval.  Ten by se na hlavní trubku asi měl nasazovat až po tom, co budou vyvrtané ostatní díry.  Pozice přípravku  se bude odvíjet od děr vyvrtaných pro přední spojky.

Elizaa62, 2018-05-18:
Mohl bys mi to nějak prosím vysvětlit? Nerozumím vůbec tomu, co mám dělat.

kaklik, 2018-05-18:
V podélné trubce je v přední části, tam kde je podložka docela hodně děr, které musí být všechny v přesných vzdálenostech a navíc souosé. Na tyto díry se montují různé díly, držák motoru, spojky, podložka a podobně. 
Potřeboval bych jednu vrtací šablonu, která se nasune na trubku a jenom se podle ní všechny díry vyvrtají. 

Teď je to vyřešené tak, že máme několik vrtacích přípravků, což je hodně užitečné na opravy, ale na výrobu je to hrozně pracné.

Elizaa62, 2018-05-23:
Vytvořila jsem nový díl 888_1014. Je to začátek svrtávacího přípravku.

kaklik, 2018-05-23:
Ano, je to správně.  Nemusí tam ale být ta celá část pro vertikální trubku. Stačí jenom ty otvory na šrouby pro spojku.

kaklik, 2018-06-11:
Všiml jsem si, že v přední části jsou otvory kolmo na ostatní otvory v hlavní trubce. 

[image]

To je správně? 

Potom přípravek neobsahuje dělení, aby šel vytisknout v omezených tiskových rozměrech.

Elizaa62, 2018-06-11:
Tohle ještě není hotové.

Elizaa62, 2018-06-13:
Opraveno. Přidala jsem dělení pro tisk.

closed 2018-09-10 by kaklik